ome/bioformats #3454: Support for Philips TIFF

fireofearth, 2019-10-22:
It appears that Philips TIFF is not supported by Bio-Formats (as of v 6.2.1):

Table 1 of *Bringing Open Data to Whole Slide Imaging*
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6774793/

Supported Formats
https://docs.openmicroscopy.org/bio-formats/6.2.1/supported-formats.html

I've tried:

- Viewing Philips TIFF directly. That's not possible using OMERO.iviewer, OMERO.insight or QuPath with Bio-Format backend

- Converting Philips TIFF to OME-TIFF using
`bfconvert -compression JPEG -bigtiff -overwrite VOA-TEST.tiff VOA-TEST.ome.tiff`
The converted images are unable to be viewed via OMERO.iviewer, OMERO.insight or QuPath with OpenSlide or Bio-Format backends

Virtually all the WSI my lab is working with are compressed Philips TIFF of 1-4G and 40-50G uncompressed and ~6 levels pyramidal (> 600 of them). Currently we use OpenSlide as a backend for viewers (i.e. QuPath, https://github.com/openslide/openslide-python)

fireofearth, 2019-10-22:
I would really suggest this as doctors who use Philips products will be producing WSIs in Philips TIFF, and usually large companies are slow to adopt new standards.

fireofearth, 2019-10-22:
If needed I can supply debuggin output from QuPath, etc.

codybum, 2020-05-18:
I second this!  We don't have good methods to produce reasonably sized OME-TIFFS from Philips TIFFs.

dgault, 2020-05-20:
Thanks for the feedback, it is certainly a format that has been requested multiple times.

Our position on adding new formats currently is better outlined in our blog post (https://blog.openmicroscopy.org/community/file-formats/2019/06/25/formats/).

We have had companies such as 3i, who have taken over support of their own file format and other companies have added support for their format either by contributing directly to the library or by commissioning Glencoe Software to do so. Where necessary, the open source team has added support for formats that are needed for their funded priorities like datasets published in the Image Data Resource.