ome/bioformats #3461: bfconvert and showinf 6.3.0 issues

ebremer, 2019-11-04:
1) bvconvert issue
D:\vsi>bfconvert -series 7 -tiley 512 -tiley 512 -compression JPEG -crop 32256,8704,5120,5120 001738-000598_01_01_2019071702.vsi ha.tif
001738-000598_01_01_2019071702.vsi
CellSensReader initializing 001738-000598_01_01_2019071702.vsi
[CellSens VSI] -> ha.tif [Tagged Image File Format]
Tile size = 512 x 512
Exception in thread "main" loci.formats.FormatException: X:32256 must be < 5120
        at loci.formats.FormatWriter.checkParams(FormatWriter.java:453)
        at loci.formats.out.TiffWriter.saveBytes(TiffWriter.java:220)
        at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:878)
        at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:763)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:691)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:1051)

The image in that series is 49512x38578.
The command line option "-crop" says:
 -crop: crop images before converting; argument is 'x,y,w,h'

2) showinf issue
The command line option "-series" says:
-series: specify which image series to read

If I leave off -series on say my above vsi image, all 16 series info shows.  If I put a -series 7, 16 series will be displayed.  Shouldn't it only show the info for series 7?

sbesson, 2019-11-05:
@ebremer answering the `showinf` question, the first step of the`showinf` utility is to initialize the reader. This will detected all series/resolutions associated with the fileset and display them int the command-line output. The `-series` option only affects the image which is displayed by the image viewer.

Thanks for reporting the issue on the conversion command, I can reproduce it with the Bio-Formats 6.3.0 command-line utility and the following minimal command matching your dimensions `bfconvert -crop 32256,8704,5120,5120 "test&sizeX=49512&sizeY=38578.fake" test.tif`. More investigation is required to know where the source of the bug lies and we will update this issue accordingly.

melissalinkert, 2019-11-11:
https://github.com/ome/bioformats/pull/3463 proposes a fix for the ```bfconvert``` issue.

PR #3463 (2019-11-29): bfconvert: fix the writer's XY coordinate when the input image is cropped
Fixes #3461.  The previous behavior was to use the same XY coordinate when reading and writing a tile.  This is fine when there is no cropping, but otherwise results in a coordinate that fails the writer's bounds check.

Without this PR, ``` bfconvert -crop 32256,8704,5120,5120 "test&sizeX=49512&sizeY=38578.fake" test.tif``` should throw:

```
Exception in thread "main" loci.formats.FormatException: X:32256 must be < 5120
	at loci.formats.FormatWriter.checkParams(FormatWriter.java:453)
	at loci.formats.out.TiffWriter.sav [... cut]
closed 2019-11-29 by dgault