ome/bioformats #3462: JPEG codec broken (6.3.0)

ebremer, 2019-11-04:
bfconvert -series 2 -tiley 512 -tiley 512 -compression LZW image.vsi lzw.tif

Image looks like:
[image]

bfconvert -series 2 -tiley 512 -tiley 512 -compression JPEG image.vsi jpeg.tif

Image looks like:
[image]

sbesson, 2019-11-05:
@ebremer I failed to reproduce the error you mentioned above using either RGB images or sample VSI files. In both cases JPEG-compressed or LZW-compressed TIFF looked identical after reopening. Do you have a sample file allowing to reproduce the issue?

ebremer, 2019-11-05:
Let me try it on some other images.  The above image isn't something I can share at this time.

ebremer, 2019-11-06:
I checked the JPG compressed image I created with showinf and it looks fine.  It doesn't look fine in GIMP, Photoshop, nor windows image viewer.  GIMP actually reports "Improper JPEG sampling factors 2,2.  Apparently should be 1,1.

ebremer, 2019-11-06:
I also tried this on some random images that came out of vips and the command will work for LZW creating viewable images in all programs, but using JPEG will make images only viewable in showinf.

ebremer, 2019-11-06:
Try using https://upload.wikimedia.org/wikipedia/commons/2/2d/Chameleons_Large_Image.jpg
with:
bfconvert -series 0 -compression JPEG -tilex 256 -tiley 256 cham.jpg boo.tif
and
bfconvert -series 0 -compression LZW -tilex 256 -tiley 256 cham.jpg boo.tif

both will be viewable in showinf, but only LZW works with other programs like Photoshop and GIMP, but the JPEG compression will fail in non-showinf programs.

sbesson, 2019-11-08:
Thanks for the additional troubleshooting, it feels like this is the same underlying problem as the one reported in https://github.com/ome/bioformats/issues/3372. Proposing to close this as duplicate and stick to the former issue for tracking this bug.

ebremer, 2019-11-08:
Okay.

closed 2019-11-08 by ebremer