ome/bioformats #3464: A potential NullPointerException bug.

JulyChen728, 2019-11-14:
Hi all,

Our tool has a potential NPE bug.

It is possible that the arguments of ```row``` and ```cols``` in method [```makeGrid```](https://github.com/ome/bioformats/tree/v6.3.0/components/bio-formats-plugins/src/loci/plugins/config/SpringUtilities.java#L99) are all 0. Then, the local variable ```max``` will be 0 at line [116](https://github.com/ome/bioformats/tree/v6.3.0/components/bio-formats-plugins/src/loci/plugins/config/SpringUtilities.java#L116). Then, the loop body of the statement ```for (int i = 0; i < max; i++) {``` (at line [146](https://github.com/ome/bioformats/tree/v6.3.0/components/bio-formats-plugins/src/loci/plugins/config/SpringUtilities.java#L146)) won't be executed, leaving ``` lastCons``` to null. Finally, the variable is used by the following code at line [171](https://github.com/ome/bioformats/tree/v6.3.0/components/bio-formats-plugins/src/loci/plugins/config/SpringUtilities.java#L171):
```
pCons.setConstraint(SpringLayout.SOUTH, Spring.sum(Spring.constant(yPad),
      lastCons.getConstraint(SpringLayout.SOUTH)));
```
Thanks.

sbesson, 2019-11-18:
Thanks for the report @JulyChen728, this particular API is not extensively used by most Bio-Formats tools which might explain how this use case went uncatched.

I assume that in the case where `lastCons` was not set as per the conditions described above, the the `setConstraint` calls should be skipped, is that correct? How confident would you feel about opening a Pull Request adding a nullity check in the relevant line of the code?

JulyChen728, 2019-11-20:
> Thanks for the report @JulyChen728, this particular API is not extensively used by most Bio-Formats tools which might explain how this use case went uncaught.
> 
> I assume that in the case where `lastCons` was not set as per the conditions described above, the the `setConstraint` calls should be skipped, is that correct? How confident would you feel about opening a Pull Request adding a nullity check in the relevant line of the code?

I have created a PR at #3469.

closed 2019-11-21 by sbesson