ome/bioformats #3472: -series argument not working

mmoisse, 2019-11-24:
When running bftools with the -series command I get the following error:

```
/software/bftools/bftools_6.3.0/bfconvert -tilex 256 -tiley 256 -crop 0,0,256,256 -series 17 ./BM.vsi split_BM/split_BM_256x256_S%s_N%n_C%c_W%w_Z%z_T%t_X%x_Y%y_M%m_A%A.tiff
./BM.vsi
CellSensReader initializing ./BM.vsi
[CellSens VSI] -> split_BM/split_BM_256x256_S%s_N%n_C%c_W%w_Z%z_T%t_X%x_Y%y_M%m_A%A.tiff [Tagged Image File Format]
Tile size = 256 x 256
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 17, Size: 1
        at java.util.ArrayList.rangeCheck(ArrayList.java:653)
        at java.util.ArrayList.get(ArrayList.java:429)
        at ome.xml.model.OME.getImage(OME.java:698)
        at ome.xml.meta.OMEXMLMetadataImpl.setPixelsSizeX(OMEXMLMetadataImpl.java:12066)
        at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:839)
        at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:763)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:691)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:1051)
Command exited with non-zero status 1

```

I believe this has to do with the following line of code, which resets the image array to 0 elements, while at the error a array of size `serie` is expected

https://github.com/ome/bioformats/blob/02223b736d93f3a02164a5c34895c69e16b0f658/components/bio-formats-tools/src/loci/formats/tools/ImageConverter.java#L536-L538

dgault, 2019-11-25:
Hi @mmoisse, thank you for reporting the issue. Would you be able to upload the sample file you were using to https://www.openmicroscopy.org/qa2/qa/upload/ for testing?

mmoisse, 2019-11-25:
Dear @dgault it is a 10GiB vsi file, so it might not be the best idea.. maybe I can try to replicate it on test vsi file? Might you known where I could find one?

dgault, 2019-11-25:
If needs be we can arrange ftp transfer for larger files but if it can be reproduced with a smaller sample that would be great. We dont have any public vsi datasets but I would expect the behaviour to be the same for other formats, our public sample data is available at https://downloads.openmicroscopy.org/images/.

Has the issue only occurred with version 6.3.0 or have you seen it with older versions?

mmoisse, 2019-11-25:
I first tried version 5.3.4 where I had a similar issue

```
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 17, Size: 1
        at java.util.ArrayList.rangeCheck(ArrayList.java:657)
        at java.util.ArrayList.get(ArrayList.java:433)
        at ome.xml.model.OME.getImage(OME.java:696)
        at ome.xml.meta.OMEXMLMetadataImpl.setPixelsSizeX(OMEXMLMetadataImpl.java:12066)
        at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:706)
        at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:630)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:559)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:880)

```
I could recapitulate the error with this file https://downloads.openmicroscopy.org/images/Ventana/openslide/OS-1.bif

```
/software/bftools/bftools_6.3.0/bfconvert -tilex 256 -tiley 256 -crop 0,0,256,256 -series 2 ./OS-1.bif split_OS-1/split_OS-1_256x256_S%s_N%n_C%c_W%w_Z%z_T%t_X%x_Y%y_M%m_A%A.tiff
./OS-1.bif
VentanaReader initializing ./OS-1.bif
Reading IFDs
Populating metadata
Populating OME metadata
[Ventana .bif] -> split_OS-1/split_OS-1_256x256_S%s_N%n_C%c_W%w_Z%z_T%t_X%x_Y%y_M%m_A%A.tiff [Tagged Image File Format]
Tile size = 256 x 256
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 2, Size: 1
        at java.util.ArrayList.rangeCheck(ArrayList.java:657)
        at java.util.ArrayList.get(ArrayList.java:433)
        at ome.xml.model.OME.getImage(OME.java:698)
        at ome.xml.meta.OMEXMLMetadataImpl.setPixelsSizeX(OMEXMLMetadataImpl.java:12066)
        at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:839)
        at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:763)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:691)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:1051)
Command exited with non-zero status 1

```

dgault, 2019-11-28:
Thanks @mmoisse, I was able to reproduce with the recommended sample file.

This seems to occur when using the specific below combination of parameters:
- series which is not the first
- tilex and tiley
- crop
- pattern output with %x and %y or %m

Removing the pattern from the output file will work as a short term workaround but I should be able to have a fix in place for this in an upcoming patch release.

closed 2019-12-12 by sbesson