ome/bioformats #3473: -crop fails when image contains multipe series of different size

mmoisse, 2019-11-25:
When using the `-crop` command on files with multiple series of varying size bftools crashed when reaching smaller sized series.

```
/software/bftools/bftools_6.3.0/bfconvert -tilex 256 -tiley 256 -crop 0,0,256,256 ./OS-1.bif split_OS-1/split_OS-1_256x256_S%s_N%n_C%c_W%w_Z%z_T%t_X%x_Y%y_M%m_A%A.tiff
./OS-1.bif
VentanaReader initializing ./OS-1.bif
Reading IFDs
Populating metadata
Populating OME metadata
[Ventana .bif] -> split_OS-1/split_OS-1_256x256_S%s_N%n_C%c_W%w_Z%z_T%t_X%x_Y%y_M%m_A%A.tiff [Tagged Image File Format]
Tile size = 256 x 256
Switching to BigTIFF (by file size)
        Series 0: converted 1/1 planes (100%)
Tile size = 256 x 256
        Series 1: converted 1/1 planes (100%)
Tile size = 256 x 256
        Series 2: converted 1/1 planes (100%)
Tile size = 256 x 256
        Series 3: converted 1/1 planes (100%)
Tile size = 256 x 256
        Series 4: converted 1/1 planes (100%)
Tile size = 256 x 256
        Series 5: converted 1/1 planes (100%)
Tile size = 256 x 256
        Series 6: converted 1/1 planes (100%)
Tile size = 256 x 256
        Series 7: converted 1/1 planes (100%)
Tile size = 256 x 256
        Series 8: converted 1/1 planes (100%)
Tile size = 256 x 256
Exception in thread "main" loci.formats.FormatException: Invalid tile size: x=0, y=0, w=256, h=256
        at loci.formats.FormatTools.checkTileSize(FormatTools.java:1026)
        at loci.formats.FormatTools.checkPlaneParameters(FormatTools.java:1002)
        at loci.formats.in.VentanaReader.openBytes(VentanaReader.java:205)
        at loci.formats.FormatReader.openBytes(FormatReader.java:878)
        at loci.formats.ImageReader.openBytes(ImageReader.java:445)
        at loci.formats.tools.ImageConverter.getTile(ImageConverter.java:1032)
        at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:826)
        at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:763)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:691)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:1051)
Command exited with non-zero status 1

```

The image is available at https://downloads.openmicroscopy.org/images/Ventana/openslide/OS-1.bif

```
/software/bftools/bftools_6.3.0/bfconvert -version
Version: 6.3.0
Build date: 24 October 2019
VCS revision: 584ab98f489398e16e7ce8cdd7a5c80dab133cde
```

dgault, 2019-11-28:
Thanks @mmoisse, I was able to reproduce with the given sample file. This error is slightly different in that it occurs when the crop size is greater than the dimensions of one of the series.. In this case the bfconvert tool should be smarter and request a smaller size. This should be fixed in an upcoming patch release.

mmoisse, 2019-12-02:
Thanks, in the meanwhile I made a fix for it. Would it be helpful to share my code?

joshmoore, 2019-12-02:
Always! It's great to see what works for others.

closed 2019-12-04 by sbesson