ome/bioformats #3476: ND2 - Incorrect time stamps

dgault, 2019-12-03:
This issue was raised in QA-28066 and a sample file provided.

With the given sample file in NIS Viewer a timestamp is reported each second from `0.03s` to `30.03s`.

With Bio-Formats 6.3.0 the timestamps are incorrect and listed as:
```
timestamp #01	-3.2057595233010108E-227
timestamp #02	-8.20992749776759E15
timestamp #03	-6.695168056434897E-159
timestamp #04	-6.672096624461741E-72
timestamp #05	-1.6277716251055805E248
timestamp #06	3.426442023302327E-266
timestamp #07	-5.57851007390377E111
timestamp #08	6.304139403323953E217
timestamp #09	2.323617524437495E106
timestamp #10	-4.033171194282644E159
timestamp #11	-1.3113301781193157E265
timestamp #12	-1.9166398963811644E-273
timestamp #13	-1.8856976627843214E-115
timestamp #14	1.8752895348237397E-121
timestamp #15	1.682467991885098E56
timestamp #16	-1.7309124511913949E264
timestamp #17	-9.751748184272613E-283
timestamp #18	-2.0768073710026513E82
timestamp #19	1.9005193482125882E-138
timestamp #20	-4.387257306710328E-82
timestamp #21	-2.1676614291209367E-290
timestamp #22	-570103.9209080567
timestamp #23	2.1003830962135502E96
timestamp #24	-7.722088845368206E-151
timestamp #25	8.888700015742615E263
timestamp #26	-1.992098538849606E-50
timestamp #27	1.67217512598466E-218
timestamp #28	-5.446441719149626E44
timestamp #29	-1.4628320038386553E-184
timestamp #30	-2.5616407716065118E-273
timestamp #31	-1.346812346244888E228
```

dgault, 2020-01-29:
A further report of incorrect timestamps with ND2 has been received along with a sample file in QA-29043. The data in that file was originally measured with NIS 5.10. I tested it with Bio-Formats 6.3.1

melissalinkert, 2023-08-11:
Retested QA 28060, 28066, and 29043 with 7.0.0. All show correct timestamps now; https://github.com/openmicroscopy/data_repo_config/pull/588 adds to test repo to prevent future regressions.

closed 2023-08-11 by melissalinkert