ome/bioformats #3483: Issues with OperettaReader: ANSI encoding, transformation matrix

VasylVaskivskyi, 2019-12-12:
I am using **bioformats_package.jar** to convert **PerkinElmer Operetta XML** to **.OME.XML**.
The problem I am currently facing is that after conversion to **.OME.XML** I receive coordinate values transformed, probably
using transfotrmation matrix supplied in the **PerkinElmer Operetta XML**. 
Application of this transformation matrix changes distance between pictures.
For example, based on the **PositionX** coordinates I expect the distance between pictures to be **1004 px**, but after transformation the distance is **999 px**.
And **999 px** is further from optimal distance **1007 px**, that will be calculated later.

Also, while **PositionX** and **PositionY** are get transformed, **PositionZ** stays the same. 

Also there is problem with encoding.
Apparently ANSI encoding is used instead of UTF-8 in output **.OME.XML**.
This makes greek symbol "mu" in μm become "xB5".

I added a link to the google drive folder with files:
1. original PerkinElmer Operetta XML, 
2. converted to OME.XML,
3. and OME.XML with values I expected to see.

https://drive.google.com/open?id=1UEw6Os5b4tNSbxCYe4VHOqaZaTuuneGQ

dgault, 2019-12-17:
Hi @VasylVaskivskyi, you are correct that the OrientationMatrix is being applied. The logic for this can be found in https://github.com/ome/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/OperettaReader.java#L880. So I understand you correctly, do you believe this matrix is being applied incorrectly or that the matrix itself is incorrect?

For example, in the sample XML provided, what would you expect the correct logic to be for the below values:
```
      <PositionX Unit="m">-0.001501518</PositionX>
      <PositionY Unit="m">-0.004804857</PositionY>
      <PositionZ Unit="m">9.7E-05</PositionZ>
      <AbsPositionZ Unit="m">0.135341004</AbsPositionZ>
      <OrientationMatrix>[[0.994753,0,0,-11.0],[0,-0.994753,0,14.0],[0,0,1.00,0]]</OrientationMatrix>
```

For the encoding, how are you using Bio-Formats, is it part of a Java program or are you using command line tools? Are you on a Windows system?

VasylVaskivskyi, 2019-12-17:
Hi @dgault,

My OS is **Windows 10**, I used **bioformats_package.jar v6.3.0** to compile and run script that does extraction and conversion of metadata from **Perkin Elmer XML** to **OME.XML**.
This is gist of the script
https://gist.github.com/VasylVaskivskyi/436bac9bf54a08bbd8523153bd1c51c2

For these coordinates I expect them to stay exactly the same after conversion to **OME.XML**

```
<PositionX Unit="m">-0.001501518</PositionX>
<PositionY Unit="m">-0.004804857</PositionY>
<PositionZ Unit="m">9.7E-05</PositionZ>
<AbsPositionZ Unit="m">0.135341004</AbsPositionZ>
<OrientationMatrix>[[0.994753,0,0,-11.0],[0,-0.994753,0,14.0],[0,0,1.00,0]]</OrientationMatrix>
```

Here is an example.

Original data
```
<PositionX Unit="m">-0.000600607</PositionX>
<PositionY Unit="m">-0.003003035</PositionY>
<PositionZ Unit="m">9.4E-05</PositionZ>
<ExposureTime Unit="s">0.02</ExposureTime>
<OrientationMatrix>[[0.994753,0,0,-11.0],[0,-0.994753,0,14.0],[0,0,1.00,0]]</OrientationMatrix>
```

Converted to OME.XML **with** application of transformation matrix
```
<Plane DeltaT="0.0" DeltaTUnit="s" ExposureTime="0.02" ExposureTimeUnit="s" 
PositionX="-11.000597455615072" PositionXUnit="m" 
PositionY="14.002987278075356" PositionYUnit="m" 
PositionZ="9.4E-5" PositionZUnit="m" 
TheC="0" TheT="0" TheZ="0"/>
```
Converted to OME.XML **without** application of transformation matrix
```
<Plane DeltaT="0.0" DeltaTUnit="s" ExposureTime="0.02" ExposureTimeUnit="s" 
PositionX="-6.00607E-4" PositionXUnit="m" 
PositionY="-0.003003035" PositionYUnit="m" 
PositionZ="9.4E-5" PositionZUnit="m" 
TheC="0" TheT="0" TheZ="0"/>
```

I expect data to look like in the last case, i.e. **without** application of transformation matrix.
In case of applying transformation matrix 
- distance between captured neighbouring images is changed
- objective position values (PositionX, PositionY) become weird: **X** -11 m, **Y** 14 m and **Z** 94 um, instead of **X** 6 mm, **Y** 3 mm, and **Z** 94 um.

\
Here is example from another dataset, but same microscope.

Original data
```
<PositionX Unit="m">-0.002454094</PositionX>
<PositionY Unit="m">0.012270469</PositionY>
<PositionZ Unit="m">2E-06</PositionZ>
<ExposureTime Unit="s">0.1</ExposureTime>
<OrientationMatrix>[[0.998534,0,0,-49.3],[0,-0.998534,0,-84.3],[0,0,1.33,28.266]]</OrientationMatrix>
```

Converted to OME.XML **with** application of transformation matrix
```
<Plane DeltaT="0.0" DeltaTUnit="s" ExposureTime="0.1" ExposureTimeUnit="s" 
PositionX="-49.30245049629819" PositionXUnit="m" 
PositionY="-84.31225248049245" PositionYUnit="m" 
PositionZ="28.266002659999998" PositionZUnit="m" 
TheC="0" TheT="0" TheZ="0"/>
```

Converted to OME.XML **without** application of transformation matrix
```
<Plane DeltaT="0.0" DeltaTUnit="s" ExposureTime="0.1" ExposureTimeUnit="s" 
PositionX="-0.002454094" PositionXUnit="m" 
PositionY="0.012270469" PositionYUnit="m" 
PositionZ="2.0E-6" PositionZUnit="m" 
TheC="0" TheT="0" TheZ="0"/>
```

Here you can see that applying transformation matrix makes objective position to go even further away: **X** -49 m, **Y** 84 m, **Z** 28 m, instead of **X** 0.2 cm, **Y** 1.2 cm and **Z** 2 um. 

**My conclusion is that this transformation or orientation matrix should not be applied to XYZ coordinates.**

dgault, 2021-11-22:
Hi @VasylVaskivskyi, we have now tested and merged changes included in https://github.com/ome/bioformats/pull/3734
This will be released in the next month as part of Bio-Formats 6.8.0 and should help with the transformation matrix issues

closed 2025-06-29 by sbesson