ome/bioformats #3487: ND2 - Timelapse stacks open as single image

mfkeles, 2019-12-26:
I've been trying to open .nd2 files with bioformats. The files open but there is only a single frame in each file. If I try to open the same file with NIS-Viewer or NIS Advanced Research I can see the timelapse. 

Has anyone encountered this issue before?

melissalinkert, 2019-12-27:
Thank you for reporting this, @mfkeles.  Are you using Bio-Formats in ImageJ, or something else?  Are you using the latest version of Bio-Formats (6.3.1)?  If not, does the same problem occur after upgrading to 6.3.1?

mfkeles, 2019-12-30:
Yes, I am using Bio-Formats in ImageJ. It's the latest version (6.4.0). I noticed that this issue occurs only at high FPS recordings that are binned 4X4. I tested some files with 2X2 binning and lower frame rates and had success. However, for 4x4 binned files, I can only see the first frame. Even though the files are large and NIS-Viewer opens them as a stack.

dgault, 2020-01-06:
Hi @mfkeles, if you are able to upload a sample file displaying this issue for us to test that would be very helpful. You can upload the file to https://www.openmicroscopy.org/qa2/qa/upload/

mfkeles, 2020-01-06:
Hi @dgault ! I've uploaded the file. Here's the info:

QA Bug | 29006 (new)

Filename: Voltron_Test004.nd2

The file opens as a timelapse in NIS-Viewere but opens as a single image in ImageJ. 

Thank you!

dgault, 2023-05-02:
Another report of a similar issue was received via the nightade-support mailing list. In that example the file should have 18 timepoints and 9 z slices. The reader has detected the tag `CustomData|Z`, and when attempting to parse the Z positions it has found 18 unique z positions. At [this point](https://github.com/ome/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/NativeND2Reader.java#L1587) the reader recognises that the number of unique positions equals sizeT and tries to be smart and sets T back to 1 and adds new series instead.

melissalinkert, 2023-08-07:
Using Bio-Formats 7.0.0, the file in QA 29006 shows 20000 timepoints as expected based on NIS Elements Viewer. Closing as fixed.

I didn't see an obvious file in `inbox` corresponding to https://github.com/ome/bioformats/issues/3487#issuecomment-1531697684, but feel free to reopen if there is a file somewhere.

closed 2023-08-07 by melissalinkert