ome/bioformats #3490: ND2 Incorrect getBitsPerPixel (Bit Depth)

MXGHarryLiu, 2020-01-06:
Hi bioformats developers, 

I notice a possible bug with **6.3.1**. For the sample files, please refer to QA **29008**. I am using MATLAB r2019b. 
To reproduce the issue, run the following scripts on the two files. 
```
reader = bfGetReader(path)
x = reader.getBitsPerPixel
```
Where `path` is the path of the file. 
The file 'E12-2p_1-1.nd2' will yield anticipated value: `x = 12`, whereas 'XMAS1_GFP.nd2' will be incorrect with `x = 16`. 
I also include a screenshot of NIS Viewer confirming that the latter recording is actually 12-bit. 
[image]
Let me know if you need any information from me. 

Thanks,

dgault, 2020-01-07:
Hi @MXGHarryLiu, thanks for the report and the provided sample file. When Bio-Formats reads the file it has 2 different metadata values stored within it:
```
uiBpcInMemory	16
uiBpcSignificant	12
```

Bio-Formats will read the Pixels as 16bit but both of these values will be able to be read using
```
bitsPerPixel = reader.getMetadataValue("uiBpcInMemory");
signifigantBits = reader.getMetadataValue("uiBpcSignificant");
```

MXGHarryLiu, 2020-01-07:
Hi @dgault, 
Thanks for the prompt reply. I realized that the bit depth of the data and the actual data type to represent the data might be different. Does this mean that the file saves 12 bit info using 16 bit encoding? Besides using the `getMetadataValue` function, is there a way we can distinguish the two (like a counterpart get function), and please remind me that which bit depth is `getBitsPerPixel` referring to? What relevant to us might be the actual bit depth. 
Do you think I can use 
```
min(reader.getMetadataValue("uiBpcInMemory"), reader.getMetadataValue("uiBpcSignificant"))
``` 
to get what we want?

dgault, 2020-01-10:
Yes essentially each Pixel will be stored in a 16bit data type even if the actual pixel data only has 12  bits of useful values. The `getBitsPerPixel` will refer to the number of bits which Bio-Formats is actually using to store each Pixel and so will refer to the data type rather than the number of significant bits.

The function you outlined would work fine to retrieve the value you want, or if you are only ever dealing with ND2 files of this nature then simply reading reader.getMetadataValue("uiBpcSignificant") would likely be enough.

MXGHarryLiu, 2021-03-31:
Issue persists with **v6.6.1** with MATLAB r2021a. Adopted the workaround as suggested.

closed 2021-03-31 by MXGHarryLiu