ome/bioformats #3502: ND2: Incorrect dimensions

dgault, 2020-01-24:
Issue was raised in the thread https://forum.image.sc/t/bioformats-reads-nd2-file-series-incorrectly/33263 and a sample file is provided in QA-29035

I have been able to reproduce the issue as described with Bio-Formats 6.3.1
The incorrect planes can be spotted by comparing the planes with the correct ordering in NIS Viewer. None of the ND2 configuration options have any impact on the result.

dgault, 2020-02-20:
A further example of this issue has been reported in forum thread https://forum.image.sc/t/hyperstack-ordering/34041

A further sample file was provided in QA-29075 and the bug was reproduced using Bio-Formats 6.3.1

In this case the ordering of the positions is incorrect and results in an ordering similar to the below:
```
BF                                                     Expected
Series 1, Channel 1, Timepoint 1 -> Position 1, Channel 1, Timepoint 1 
Series 1, Channel 1, Timepoint 2 -> Position 2, Channel 1, Timepoint 1 
Series 1, Channel 1, Timepoint 3 -> Position 3, Channel 1, Timepoint 1 
Series 1, Channel 1, Timepoint 4 -> Position 1, Channel 1, Timepoint 2
Series 1, Channel 1, Timepoint 5 -> Position 2, Channel 1, Timepoint 2
Series 1, Channel 1, Timepoint 6 -> Position 3, Channel 1, Timepoint 2
```

Parm0n, 2021-06-10:
@dgault Hi David, 
I'm trying to solve this problem, but I have only .nd2 documents that do not work correctly. Could you provide public access to some document (no matter what format), which would have timesteps and imaged correctly. That would help me very much to understand this bug and fix it.

sbesson, 2021-06-14:
@Parm0n some of our sample ND2 files used for the non-regression builds are available [publicly here](https://downloads.openmicroscopy.org/images/ND2/). IDR has also published studies in this format e.g. [idr0095](https://idr.openmicroscopy.org/search/?query=Name:idr0095)  or [idr0109](https://idr.openmicroscopy.org/search/?query=Name:idr0109) and the raw data can be downloaded using [Aspera](https://idr.openmicroscopy.org/about/download.html).

Parm0n, 2021-06-30:
Hi @sbesson , Thanks for sharing the ND2 files. I think we were able to solve the problem related to incorrectly read positions and timepoints, but I'm not completely sure that this was the bug that is being solved in this conversion :) . Could you please provide access, if possible, to the files that are causing the import problem for you, so that I can test them.

Hi @dgault, I have a question for you regarding the OMERO.web application, which may also be related to the problem discussed here. I have a screenshot of the OMERO.web application, which shows that some .nd2 document is open. This file imaged  as a grid in which the parts of this document are located, in addition, below you can see a separate view for the fields from well. 
[image]

I also found this on the application documentation page (https://omero-guides.readthedocs.io/en/latest/introduction/docs/data-management.html     Step-by-Step -> Browsing and rendering -> Point number 9).

The question is how can I see this interface for well plates?
Could you please provide access to such a file, which could be displayed in the OMERO.web in the same way as in the screenshot. Or write step by step if I need to change the application settings in some special way in order to see such an interface.

Thank you :)

dgault, 2021-07-01:
Support for Screens and Plates is on a format by format basis and currently is not supported for ND2 data. If you want to check if it is supported for a specific format you should be able to see via the supported metadata page for the format. ND2: https://docs.openmicroscopy.org/bio-formats/6.6.1/metadata/NativeND2Reader.html for example compared to Perkin Elmer Columbus which supports Screen and Well metadata: https://docs.openmicroscopy.org/bio-formats/6.6.1/metadata/ColumbusReader.html

SimoneNikon, 2021-07-12:
Hi dgault, we are aware that support for well plates is currently missing for nd2, but we would really like to change this. The information from nd2 side is there as we have well plate views within NIS elements, we just need to make it available to you. Can you help us with an advise for the best way how to do this? Thanks a lot for your help :)

dgault, 2021-07-12:
Hi @SimoneNikon, thank you for reaching out. If you are able to link to or provide an existing specification or documentation on how to identify the relevant metadata within the format that would be a great help. If you would need to share privately you can submit it using https://www.openmicroscopy.org/qa2/qa/upload/. Alternatively if developers wish to open a PR to add support for plate views I can help point to the correct API calls required, a good example of how to populate the required metadata would be https://github.com/ome/bio-formats-examples/blob/master/src/main/java/FileExportSPW.java#L166 or https://github.com/ome/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/MetamorphReader.java#L861-L893

closed 2021-09-29 by dgault