ome/bioformats #3510: SCN: Incorrect image color

dgault, 2020-02-19:
This issue was first raised on forum thread: https://forum.image.sc/t/leica-scn-images-with-wrong-colour/34077/2

SCN files have been reported to import with a pink background with a grey colour as below:
[image]
[image]

Two sample files have been provided in QA-29074
These have been tested with Bio-Formats 6.3.1 and the issue was reproduced as described in the original thread.

This is similar to issues which we have seen and fixed previously on this format (see https://github.com/ome/bioformats/pull/3240)

Jcarlosmiguel, 2020-02-24:
I also can find the problem when using omero insight and omero server 5.6

Jcarlosmiguel, 2020-04-28:
Any upgrades on this? Is there a way that I can circunvet the problem converting the images?

dgault, 2020-04-30:
Hi @Jcarlosmiguel, Im afraid we do not yet have a fix in the Leica SCN Reader for this issue. Do you have the option to export or convert them to an other format?

Jcarlosmiguel, 2020-04-30:
Unfortunately I don’t have that option. But I could use an older version of OMERO in a Virtual Machine and import to it. Then export  in another format. I never done that so any help or suggestion is welcome. Do you know which was the latest OMERO version fixed?

dgault, 2020-05-01:
Unfortunately this files will still show the same issue in older systems. 

Do you know of any specific differences for which files show the issue? For example we currently disable color correction for newer Versa systems which we base on the dev model in the metadata. The uploaded sample images are from `Leica SCN400`, do you know if its only files from this specific system for example?

Jcarlosmiguel, 2020-05-07:
I only have Leica SCN400 and Hamamatsu files. The Hamamatsu files are fine.

dgault, 2020-05-14:
I have opened a PR to test a potential fix (https://github.com/ome/bioformats/pull/3562). The issue will be if this bug is present in all Leica SCN400 files or if there is a variation within that system.

Jcarlosmiguel, 2020-06-19:
I discovered this is true for bio-formats loader. If using QuPath I can inport using openslide without issues