ome/bioformats #3511: LIFReader fails on magic bytes in lof file

mtbc, 2020-02-24:
On `Image017_pmd_0.lof` from https://www.openmicroscopy.org/qa2/qa/feedback/29089/ Bio-Formats is giving me,
```
LIFReader initializing /home/mtbc/ome/apache_repo/29089/Image017_pmd_0.lof
loci.formats.in.LIFReader.initFile(/home/mtbc/ome/apache_repo/29089/Image017_pmd_0.lof)
Reading header
Finding image offsets
Looking for a block at 43; 0 blocks read
Location.mapFile: embedded-stream.raw -> null
Location.mapFile: embedded-stream.raw -> null
Failure during the reader initialization

loci.formats.FormatException: Invalid Memory Block: found magic bytes 298, expected 112
	at loci.formats.in.LIFReader.initFile(LIFReader.java:523) ~[bioformats_package.jar:6.4.0-SNAPSHOT]
	at loci.formats.FormatReader.setId(FormatReader.java:1389) ~[bioformats_package.jar:6.4.0-SNAPSHOT]
	at loci.formats.ImageReader.setId(ImageReader.java:843) ~[bioformats_package.jar:6.4.0-SNAPSHOT]
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650) ~[bioformats_package.jar:6.4.0-SNAPSHOT]
	at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1035) [bioformats_package.jar:6.4.0-SNAPSHOT]
	at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121) [bioformats_package.jar:6.4.0-SNAPSHOT]
```
I also notice that `lof` is not listed as a filename extension on https://docs.openmicroscopy.org/bio-formats/6.3.1/supported-formats.html.

melissalinkert, 2020-02-24:
LIF and LOF are two separate formats with similar headers.  LOF is not yet supported: https://trello.com/c/C13JeiIu/92-add-support-for-leica-xlef-xllf-lof-files

mtbc, 2020-02-24:
Aha, perhaps simply the issue is that the LIF reader is trying to pick it up at all, thank you.

mtbc, 2020-02-26:
For the record, the image I reference was taken with a Leica TCS SP8 confocal.

closed 2022-02-16 by dgault