ome/bioformats #3513: Leica LIF: Missing laser settings

dgault, 2020-02-25:
Issue was raised on forum thread https://forum.image.sc/t/leica-lif-metadata-fiji-extracting-laser-settings-for-multiple-channels/34250/4

Device information is correctly populated on OME-XML
However additional metadata such as laser intensity shown in 'Show Info' only contains values for a single laser. This may be because the same key values are being used for each laser and overwriting the previous entries. Further investigation is required to confirm.

A sample file has been made available in QA-29127 and issue reproduced with Bio-Formats 6.3.1

dgault, 2020-08-20:
It has also been reported on the same image sc thread that this also impacts other per-channel metadata such as exposure.

dgault, 2022-01-14:
I haven't gone through this in detail, and although its a slightly different sceanrio, we have encountered issues with series metadata having the same key names across each series. In these cases we append a prefix for each series (such as "Series0 |" ). This is done using a helper method to merge the full list of original metadata and the series data, appending a different prefix each time (https://github.com/ome/bioformats/blob/develop/components/formats-api/src/loci/formats/FormatReader.java#L1462). 

Perhaps it is possible to use the same arrays from the OMEXML parsing and use the MetadataTools.merge helper function to merge them into the original metadata with a different prefix such as Laser0 etc.

For the DetectionRange values, I opened up the sample file you provided and the raw metadata seems to have the values set to 0:
```
< D e t e c t o r   N a m e = " P M T   2 "   T y p e = " P M T "   S c a n T y p e = " I n t e r n a l "   C h a n n e l = " 2 "   C h a n n e l N a m e = " C h a n n e l   2 "   I s A c t i v e = " 1 "   S T E D D e t e c t o r M o d e = " 1 "   S T E D D e t e c t o r M o d e N a m e = " C o n f o c a l "   S T E D _ L a t e r a l S a t u r a t i o n F a c t o r = " 0 "   G a i n = " 3 9 9 . 9 9 6 1 8 5 2 4 4 5 2 6 "   O f f s e t = " 0 "   I s S T E D D e t e c t o r = " 1 "   I s S R S D e t e c t o r = " 0 "   D e t e c t i o n R a n g e B e g i n = " 0 "   D e t e c t i o n R a n g e E n d = " 0 "   P h a s e = " 0 "   I s F l i m D e t e c t o r = " 0 "   I s H P D D e t e c t o r = " 0 "   C a n D o T i m e G a t e = " 0 " / >
```

I will need to take a further look at the `translateLaserLines` to see exactly why the index was set to start at 2 the way it was.

zacsimile, 2022-01-14:
OK, so actually swapping `addSeriesMeta(key.toString() + attr.getName(), attr.getValue());` for addSeriesMetaList(key.toString() + attr.getName(), attr.getValue()); on line 1280 solves it. However, this now has the opposite problem of creating duplicates (one per channel).

So this looks good

[image]

But we also get a lot of unnecessary repeats

[image]

It's probably better to have all of the entries than none of them, but probably we could remove duplicates where necessary. 

Not sure what's up with the `DetectionRange` values. I have to assume these weren't set to 0 when I imaged, but maybe so? The `MultiBand` entries contain the correct wavelength I used. I guess Leica must use those instead of `DetectionRange`.

I think line 1648 is where index gets set to 2. This seems quite deliberate, and the last commit on that line is by @melissalinkert and says "Fix how laser wavelengths are linked to channels." I'm not sure what the reason for it is, but it seems well thought out. Might be worth just leaving it, unless it's causing issues.

zacsimile, 2022-06-22:
Did they upload an example .lif? Happy to have a look.

dgault, 2022-06-27:
I don't believe the example file has been uploaded yet