ome/bioformats #3517: Recent ND2 files cannot be opened by bio-formats with NegativeArraySizeException

aarpon, 2020-03-05:
Some recent ND2 files (NIS Elements version 5.2, maybe earlier; you can find an example here: https://drive.google.com/file/d/1nrLmO6KAfUVNf6Ie5YXupZZf5LQLgoBd/view?usp=sharing) fail opening in bio-formats (up to version 6.3.1) with the following exception:

```
(Fiji Is Just) ImageJ 2.0.0-rc-69/1.52p; Java 1.8.0_172 [64-bit]; Windows 10 10.0; 1798MB of 24263MB (7%)
 
java.lang.NegativeArraySizeException
	at loci.formats.in.NativeND2Reader.openBytes(NativeND2Reader.java:305)
	at loci.formats.DelegateReader.openBytes(DelegateReader.java:227)
	at loci.formats.ChannelFiller.openBytes(ChannelFiller.java:156)
	at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:229)
	at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:161)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.formats.DimensionSwapper.openBytes(DimensionSwapper.java:233)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.plugins.util.ImageProcessorReader.openProcessors(ImageProcessorReader.java:186)
	at loci.plugins.in.ImagePlusReader.readProcessors(ImagePlusReader.java:422)
	at loci.plugins.in.ImagePlusReader.readPlanes(ImagePlusReader.java:387)
	at loci.plugins.in.ImagePlusReader.readImage(ImagePlusReader.java:282)
	at loci.plugins.in.ImagePlusReader.readImages(ImagePlusReader.java:243)
	at loci.plugins.in.ImagePlusReader.readImages(ImagePlusReader.java:221)
	at loci.plugins.in.ImagePlusReader.openImagePlus(ImagePlusReader.java:116)
	at loci.plugins.in.Importer.readPixels(Importer.java:149)
	at loci.plugins.in.Importer.run(Importer.java:86)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:230)
	at ij.IJ.runPlugIn(IJ.java:193)
	at ij.IJ.runPlugIn(IJ.java:182)
	at HandleExtraFileTypes.openImage(HandleExtraFileTypes.java:499)
	at HandleExtraFileTypes.run(HandleExtraFileTypes.java:72)
	at ij.IJ.runUserPlugIn(IJ.java:230)
	at ij.IJ.runPlugIn(IJ.java:193)
	at ij.IJ.runPlugIn(IJ.java:182)
	at ij.io.Opener.openWithHandleExtraFileTypes(Opener.java:516)
	at ij.io.Opener.openImage(Opener.java:370)
	at ij.io.Opener.openImage(Opener.java:242)
	at ij.io.Opener.open(Opener.java:109)
	at ij.io.Opener.openAndAddToRecent(Opener.java:291)
	at ij.plugin.DragAndDrop.openFile(DragAndDrop.java:192)
	at ij.plugin.DragAndDrop.run(DragAndDrop.java:159)
	at java.lang.Thread.run(Thread.java:748)
```

dgault, 2020-03-06:
Thanks for reporting the bug and providing a sample file. I was able to reproduce the exception using the latest Bio-Formats release (6.3.1) and neither of the ND2 options (chunkmap or native lib) had any impact.

Would it be possible to make the provided sample file available publicly under the CC-BY license?

aarpon, 2020-03-09:
Hi,

sorry for the delay: I had to double-check with the owner of the file. The
file can be made available publicly under the terms of the CC-BY license.
Do I have to do something about it myself?

One more thing: we cropped the image in NIS Elements and the smaller file (
https://drive.google.com/file/d/1_qR5lagexpVq0rgkHvRAFhopBxPCL6ok/view?usp=sharing)
could be opened in Fiji successfully.

Thanks,
a2

melissalinkert, 2020-11-02:
I believe https://forum.image.sc/t/error-while-opening-nd2-file/44700 is another example of this issue.  The problem appears to be specific to files with multiple channels that are stored interleaved when the total pixel count for a plane (X * Y * channels * bits) exceeds 2 GB.  Opening smaller tiles is a work-around in the meantime, but we should be able to fix this for the case when X * Y * bits is less than 2 GB.

dgault, 2021-07-09:
An additional report of this issue comes from forum thread https://forum.image.sc/t/large-nd2-file-doesnt-open-negativearraysizeexception/54897. A new sample file reproduces the exception with Bio-Formats 6.6.1

melissalinkert, 2023-08-12:
The file from https://forum.image.sc/t/error-while-opening-nd2-file/44700 no longer has this problem; https://github.com/openmicroscopy/data_repo_config/pull/590 adds it to the test repo.

I was not able to find the other files referenced here in `inbox`, though, so leaving this open until we can confirm that they are working as well.

dgault, 2024-03-12:
A further report of this issue has been received for stitched files on the forum thread: https://forum.image.sc/t/large-nd2-file-doesnt-open-negativearraysizeexception/54897/10

dominikl, 2024-10-30:
Still happens with latest version 8.0.0, see image.sc. reference above.

```
(bf) ➜  bftools ./showinf -version
Version: 8.0.0
Build date: 24 October 2024
VCS revision: b28636177d456992bf1b7199a7ce61d8af7c020b
(bf) ➜  bftools BF_MAX_MEM=16G ./showinf ../dan-iss-stitch_000_rm05_gas1-combined_final.nd2
Checking file format [Nikon ND2]
Initializing reader
ND2Reader initializing ../dan-iss-stitch_000_rm05_gas1-combined_final.nd2
...
Reading series #0 pixel data (0-4)
Exception in thread "main" java.lang.NegativeArraySizeException: -891306166
	at loci.formats.in.ND2Reader.openBytes(ND2Reader.java:320)
	at loci.formats.FormatReader.openBytes(FormatReader.java:922)
...
```

melissalinkert, 2024-10-31:
> This issue has been mentioned on **Image.sc Forum**. There might be relevant details there:
> 
> https://forum.image.sc/t/nd2-container-file-does-not-open-with-bio-formats/104299/2

The file referenced here has now been copied to `inbox/imagesc-104299/`.

dominikl, 2025-03-19:
There's another one on image.sc: https://forum.image.sc/t/large-nd2-file-one-system-can-open-but-nobody-else/108565/ 
(Respective file was copied to `inbox/imagesc-108565/`) Looks like the same issue to me.

ehrenfeu, 2025-07-30:
Cc @lguerard @rohangirishrao @cellkai @ehrenfeu (we're also potentially suffering from this issue).

melissalinkert, 2025-07-30:
Thanks, @loicsauteur and @ehrenfeu. We're taking this out of the roadmap for 8.3.0 just due to time constraints, but will aim to pick up again for the next release.

Maybe I missed this, but had either of you uploaded data for this issue? If not, could you please put one of the affected files in the [Bio-Formats community on Zenodo](https://zenodo.org/communities/bio-formats/records)? That will help with testing any potential fixes.

ehrenfeu, 2025-07-30:
I guess @loicsauteur will be able to generate such a dataset or even use an existing one.