ome/bioformats #3525: CZI: empty planes in XT line-scan

dgault, 2020-03-13:
This issue has been reported in forum thread https://forum.image.sc/t/bioformats-wont-read-czi-linescans-properly/32428/19 and follows on from a fix in 6.4.0 https://github.com/ome/bioformats/pull/3493

A sample file has been provided and the issue reproduced as described in the thread using Bio-Formats 6.4.0

The number of lines in the file is correct but a number of the planes have only zero pixel values

chtkong, 2021-08-05:
Hi there, has any progress been made on this?

dgault, 2021-08-06:
Hi @chtkong, although there have been a number of CZI improvements in the past few releases and the upcoming release Im afraid there is not yet a fix in place for this particular issue.

dgault, 2022-09-27:
An additional instance of this issue was raised by a user of forum thread https://forum.image.sc/t/line-scans-with-many-time-points-drop-97-of-data-zeiss-microscope/72044

A sample file that reproduces the issue was provided and testing using the latest Bio-Formats 6.10.1 displays the problem as described. The majority of the line scans in the file display as zero with no data showing, this is not the case when the file is read with Zen.