ome/bioformats #3530: Relicense the matlab interface to bioformats to BSD-2-Clause?

RainerHeintzmann, 2020-03-17:
Software developments in Matlab are sometimes not open source and even if they are, they need to be distributed via the compiled Matlab version as this is the only way that such software can be used by users that do not have a Matlab licence  (the required Matlab Runtime is free!).
It would be extremely useful if the bioformats reader and writer can be used also by such software.
A common way to achieve this without violating the GPL-2+ licence is to NOT include the bioformats_package.jar file in the Matlab software and instruct the user on how to install this package. 
This strategy is facilitated by using the existing matlab bioformats interface functions. 

However, if those functions (NOT the bioformats_package.jar) are included in the software to distribute, currently the WHOLE software would need to fall under GPL-2+ (the "viral aspect" of GPL-2). Some packages solve this problem by interpreting these interface functions as not being part of the distribution as long as they are not called (e.g. by first checking if the bioformats_package.jar can be located). One can also discuss about whether it is enough to include the source code of these functions somehow in the distribution (e.g. by clicking on a "show source" button) and the rest of the software is not considered derived work and therefore does not fall under GPL-2. Yet, what is needed is a **clean solution**, which would be achieved by moving these functions into the BSD-2 branch of the code and releasing them separately as  a package under BSD-2 license.

Dear contributors to these matlab interface functions, please all agree on relicensing them under BSD-2, which would make this code generally more useful!  This will, of course, have no effect on BioFormats remaining GPL-2, it just affects the ability to include the interface to Bioformats even if it is not present in the system.

sbesson, 2020-03-18:
Cross-referencing to https://github.com/ome/bioformats/pull/3524 for the original discussion., thanks for opening this issue @RainerHeintzmann. /cc @jrswedlow @chris-allan @melissalinkert. 

NB: in order to keep this discussion constructive especially at this time, I suggest avoiding the usage of the viral analogy for the GPL license.

mkitti, 2020-07-13:
I have contributed to the MATLAB interface. I authorize my contributions to be licensed under either BSD-2 or GPL-2+

RainerHeintzmann, 2020-07-13:
Thanks a lot! This is very helpful.

RainerHeintzmann, 2020-07-13:
Of course, I also authorize my contributions (if they become part of the branch) to be under BDS-2.
@sbesson , are you also OK with this relicensing to BDS-2 (i.e. moving the folder under the BSD branch of the repository?
We then need the OK from:
@jarmond 
@snoopycrimecop
@dgault 
@melissalinkert 
@carandraug
@crisluengo

crisluengo, 2020-07-13:
I have contributed to the MATLAB interface. I authorize my contributions to be licensed under either BSD-2 or GPL-2+.

Thanks, @RainerHeintzmann , this is a really good initiative. It would make things a lot clearer!

jarmond, 2020-07-13:
I authorize my contributions to be licensed under either BSD-2 or GPL-2+.
Jon Armond

melissalinkert, 2020-07-13:
After discussion with @chris-allan and @jrswedlow, I authorize my MATLAB contributions to be relicensed as BSD-2.

sbesson, 2020-07-13:
Thank you @RainerHeintzmann for re-raising the priority of this issue. We just went through a patch release of Bio-Formats which would not have been the most suitable for this type change. Our roadmap is now moving towards  a minor release which I expect would be fine for such a permissive relicensing.

In the general sense, my understanding is that relicensing minimally requires the consent of all copyright holders. As per your comment, asking for the consent of all major contributors of the MATLAB interface feels like a very sane and solid approach. Thanks to @mkitti for taking the initiative to engage on this issue and formulate a public consent statement.

I noticed that the list of contributors mentioned in https://github.com/ome/bioformats/issues/3530#issuecomment-657384778 did not include historical authors, like Curtis who is the intial contributor this interface :). I suspect this is an artifact from the renaming of files. I tried to compile a list of contributors as exhaustive as possible using `for i in $(find matlab/ -type f); do  git log --follow --pretty=format:'%an' -- $i; done | sort -u` and I'll try to keep track of the public consent in the following list with the relevant references to the public consent:

- [ ] @bramalingam 
- [ ] @bpindelski
- [ ] @carandraug 
- [x] @crisluengo https://github.com/ome/bioformats/issues/3530#issuecomment-657386300
- [ ] @ctrueden 
- [ ] @dgault 
- [x] @jarmond  https://github.com/ome/bioformats/issues/3530#issuecomment-657387673
- [x] @mkitti https://github.com/ome/bioformats/issues/3530#issuecomment-657316360
- [x] @melissalinkert https://github.com/ome/bioformats/issues/3530#issuecomment-657584955
- [ ] @sbesson 
- [ ] @joshmoore

sbesson, 2020-07-13:
I authorize my contributions to the Bio-Formats MATLAB to be relicensed as BSD-2.

RainerHeintzmann, 2020-07-13:
>
> Thanks to all replying and consenting so quickly!
>
 Just a few more to go and we are ready :-)

joshmoore, 2020-07-13:
I authorize my contributions to the Bio-Formats MATLAB to be relicensed as BSD-2.

bpindelski, 2020-07-13:
I authorize my contributions to the Bio-Formats MATLAB to be relicensed as BSD-2.

ctrueden, 2020-07-14:
I authorize my contributions to the Bio-Formats MATLAB interface to be relicensed as BSD-2.

carandraug, 2020-07-14:
I authorise my contributions to the Bio-Formats Matlab and Octave interface so far to be released under BSD-2.

RainerHeintzmann, 2021-04-28:
Are we done with the list now? I think we need the OK by
@bramalingam
and 
@dgault
Would be great to finally do the relicensing.

mkitti, 2021-04-28:
Both are current or alumni of OME. Perhaps OME has been assigned their copyright.

dgault, 2021-04-29:
I authorize my contributions to the Bio-Formats MATLAB to be relicensed as BSD-2.

joshmoore, 2021-04-29:
Let's ping @bramalingam one more time, but otherwise, yes, his contributions here were during his time with U. Dundee.

closed 2021-07-16 by dgault

bramalingam, 2021-08-11:
Apologies for the delay. Yes I authorize my contributions to the Bio-Formats MATLAB interface to be relicensed as BSD-2.