ome/bioformats #3533: Issue with .vsi-files in QuPath

TMM-98, 2020-03-23:
Hello there,

QuPath won't open my .vsi-images, which is a unique Olympus CellSens File-format.

Interestingly, Fiji is able to open these files with the bioformat-plugin, so, as the supported formats are same, I do not quite understand the reason. The Error Report is subsequently attached
Does anyone have any idea on how to solve this issue?

Thank you so much in advance!
Thiemo

[image]

[image]

melissalinkert, 2020-03-23:
It looks like you are using QuPath 0.1.2, which does not have Bio-Formats installed by default.  Do you have the Bio-Formats extension (https://github.com/qupath/qupath-bioformats-extension) installed?  If not, does installing that extension solve the problem?

sbesson, 2020-03-23:
/cc @petebankhead

petebankhead, 2020-03-23:
If you want to open the image with QuPath v0.1.2, you do indeed need the Bio-Formats extension. Documentation is [here](https://github.com/qupath/qupath/wiki/Supported-image-formats).

However, I'd recommend trying the latest milestone release of QuPath from https://github.com/qupath/qupath/releases - this may already have resolved any issues you see (in which case *don't* install the Bio-Formats extension - it isn't needed and isn't compatible with v0.2.0). Documentation for it is [here](https://qupath.readthedocs.io/en/latest/docs/intro/formats.html).

I hope that a stable version of v0.2.0 will be available within the next month or so.

If Fiji can open the image with Bio-Formats, it is unlikely to be an issue with Bio-Formats. The [image.sc discussion](https://forum.image.sc/t/importing-a-vsi-composite-image-in-imagej-via-bioformats-plugin-how-do-i-get-a-high-resolution-macro-image/35282) is probably a better place to follow up.

sbesson, 2020-03-23:
Closing and using the image.sc thread as the single place of discussion as per https://forum.image.sc/t/importing-a-vsi-composite-image-in-imagej-via-bioformats-plugin-how-do-i-get-a-high-resolution-macro-image/35282/7

closed 2020-03-23 by sbesson