ome/bioformats #3536: Cannot compress with TiffWriter if using tiling API

mtbc, 2020-03-30:
The attached [example.zip](https://github.com/ome/bioformats/files/4404049/example.zip) is a small Java project that writes four one-plane images: with and without using the tiling API, with and without using compression. One of the four is not like the others: using the tiling API and compression together shows corrupted pixel data. Is this a bug or am I using the API wrongly?

melissalinkert, 2020-03-30:
At first look, this appears to be the same issue as the one referenced via https://github.com/ome/bioformats/pull/3500.  If you apply this patch:

```
62c62,66
<                         writer.saveBytes(0, tile, ifd, x * 128, y * 128, 128, 128);
---
>                         if (isTilingApi) {
>                             writer.saveBytes(0, tile,      x * 128, y * 128, 128, 128);
>                         } else {
>                             writer.saveBytes(0, tile, ifd, x * 128, y * 128, 128, 128);
>                         }
```

then all four images should look the same.

mtbc, 2020-03-31:
If the code passes an IFD at all then it does use one IFD instance per plane; I guess I had wrongly thought it limited to an "if you're passing IFDs at all then ..." scenario. Perhaps it's "if you're tiling, you *must* pass an IFD".

Thank you, you're right, if I use the tiling API but *still* actually pass an IFD to `saveBytes` then the image comes out fine even given compression. I'd wrongly thought that much of the point of using the tiling API was that I wouldn't have to pass an IFD with the tiling size.

Perhaps the API is just for negotiating an acceptable tiling size? Which is still useful, it can help me with https://github.com/ome/omero-downloader/issues/38.