ome/bioformats #3538: NegativeArraySizeException while opening tiff from Acapella

olatarkowska, 2020-04-02:
Hello,

I am facing NegativeArraySizeException when opening the tiff image. Is that BF issue or am I doing something wrong?

[image]

dgault, 2020-04-03:
The exception is likely due to the size of the requested plane being greater than INT.MAX.
Do you know the dimensions of the image? Also do you need to open the full image in FIJI? Would it be possible to crop part of it or open it via another tool?

olatarkowska, 2020-04-03:
Hi David,

Image is bigger then 2GPix but I was trying to crop, is that not how it should work?

dgault, 2020-04-03:
Yeah, cropping should enable it to be opened, what cropped dimensions did you choose? Does it consistently reproduce the same exception with different values?

olatarkowska, 2020-04-03:
Yes it was consistent, 1000 x 1000 :-)

dgault, 2020-04-21:
Sample file has been received, I will follow up with debugging this issue further today

dgault, 2020-04-22:
As a follow up, the issue is indeed related to the plane size being greater than INT.MAX. It is a single tiled image with the tile size being the entire image and using LZW compression. The tile size also being greater than INT.MAX is the reason for the exception even when cropping.

There is an existing special case in the TiffParser for a large single tiled uncompressed image, but the TiffParser will need updated to try to handle this special case scenario also.

olatarkowska, 2020-04-22:
> > Are you saying this would take some significant effort opposed to a small fix?
> If it requires an update to TiffParser then it will require a signifigant amount of testing and review before I can put it in a release as it could potentially impact the reading of any Tiff based formats

Please let us know

joshmoore, 2020-04-23:
What kind of deadline are you looking at, @olatarkowska? If you are talking days to even a week or two, then there is unlikely to be a release for something that could impact a majority of formats.

olatarkowska, 2020-04-23:
Thanks for the question, my deadline is beggining of May

sbesson, 2020-04-23:
If I understand well, we are talking about a TIFF file containing a single compressed, extremely large tile (> 2G). Although this is completely valid from the perspective of the TIFF specification, such a design feels very inadvisable for end users. Assumign the data is read in tiles like many modern applications do, the whole plane will effectively be loaded, uncompressed and cropped from disk for each requested tile.

From the acquisition software, is the number and/or the size of the stored tiles something that the operator has control over? If not, this might be something to feed back to the manufacturer.

In parallel to the Bio-Formats discussion and mindful of the impact of the changes on all TIFF-based formats as discussed above, the only other workaround I can think of would be to rewrite the TIFF file in tiles using reasonable size matching the processing e.g. 1024x1024. ImageMagick would be one of the first tools worth trying.

sbesson, 2020-04-28:
Updating this issue following some troubleshooting and various discussions including @joshmoore, @dgault, @melissalinkert and @chris-allan.

For reference, the TIFF details for the uploaded sample file are:

```
TIFFReadDirectory: Warning, Unknown field with tag 65200 (0xfeb0) encountered.
TIFF Directory at offset 0x4e399c9e (1312398494)
  Image Width: 36432 Image Length: 40402
  Resolution: 72, 72 pixels/inch
  Bits/Sample: 16
  Sample Format: unsigned integer
  Compression Scheme: LZW
  Photometric Interpretation: min-is-black
  Rows/Strip: 40402
  SMin Sample Value: 0
  SMax Sample Value: 3181
  Planar Configuration: single image plane
  Tag 65200: <?xml version="1.0" encoding="UTF-8" ?><Root><Arrays><Array CompressionType="LZW" Type="Image" Width="36432" Height="40402" BitsPerPixel="16" Factor="1" CompressionRate="2" /><Array CompressionType="LZW" Type="Image" Width="36432" Height="40402" BitsPerPixel="16" Factor="1" CompressionRate="2" /></Arrays></Root>
TIFF Directory at offset 0xa3ef9924 (2750388516)
  Image Width: 36432 Image Length: 40402
  Resolution: 72, 72 pixels/inch
  Bits/Sample: 16
  Sample Format: unsigned integer
  Compression Scheme: LZW
  Photometric Interpretation: min-is-black
  Rows/Strip: 40402
  SMin Sample Value: 0
  SMax Sample Value: 6299
  Planar Configuration: single image plane
```

Briefly, a 36K x 40K 16-bit plane is stored as a single LZW-compressed strip with 40402 rows. As suggested by this issue, such a binary layout completely challenge the standard Bio-Formats paradigm. Typically, loading a single 1000x1000 tile as in https://github.com/ome/bioformats/issues/3538#issuecomment-608389352 will require accessing and uncompressing a 36432x100 strip. In addition to the cost of the operation as well as the amount of memory required, this will quickly reach the Java 2GB limit for byte arrays.

As it stands, proper support of this binary layout would require a major change of the codecs API  via some sort `LZWEncodedStream`. Closing this and marking as `wontfix`.

Our best suggestion for dealing with such data in the OME environment but probably other libraries would be to convert it into another data layout. I mentioned ImageMagick above and @chris-allan will follow-up with another alternative.

closed 2020-04-28 by sbesson

chris-allan, 2020-04-28:
Over the weekend I experimented a bit with following the "raw" intermediate style as you would in the `bioformats2raw` --> `raw2ometiff` pipeline in pure Python with support for 64-bit array indexing. The result is `tiff2raw`:

 * https://github.com/glencoesoftware/tiff2raw

This is highly experimental but functional; requiring several GBs of RAM to complete successfully. glencoesoftware/raw2ometiff#21 is required to read the layout and perform the conversion to OME-TIFF.

olatarkowska, 2020-04-28:
Thanks @chris-allan I will test it, We have max 750GB of RAM