ome/bioformats #3544: IFDs not read in >4Gb Hamamatsu NDPI files

jfkotw, 2020-04-09:
I ran into an issue when reading large (>4Gb) Hamamatsu NDPI files using Bioformats. See [here](https://github.com/jfkotw/LargeNDPIBug/blob/master/src/main/java/LargeNDPIBug.java) for the code to reproduce the issue. I'm importing formats-gpl 6.4.0. I'm waiting for permission to release a double size slide that allows to reproduce the error, and will post here when I have more info. But opening the issue now, in case you have one that you could test already.

```
13:48:14.176 [main] ERROR loci.formats.tiff.TiffParser - Error reading IFD type at: 2514351386
13:48:14.176 [main] DEBUG loci.formats.tiff.TiffParser - 
loci.common.enumeration.EnumException: Unable to find IFDType with code: 8528
	at loci.formats.tiff.IFDType.get(IFDType.java:90) ~[formats-bsd-6.4.0.jar:6.4.0]
	at loci.formats.tiff.TiffParser.readTiffIFDEntry(TiffParser.java:1405) ~[formats-bsd-6.4.0.jar:6.4.0]
	at loci.formats.tiff.TiffParser.getIFD(TiffParser.java:464) ~[formats-bsd-6.4.0.jar:6.4.0]
	at loci.formats.tiff.TiffParser.getFirstIFD(TiffParser.java:391) ~[formats-bsd-6.4.0.jar:6.4.0]
	at loci.formats.in.SEQReader.isThisType(SEQReader.java:87) ~[formats-gpl-6.4.0.jar:6.4.0]
	at loci.formats.FormatReader.isThisType(FormatReader.java:612) ~[formats-api-6.4.0.jar:6.4.0]
	at loci.formats.in.SEQReader.isThisType(SEQReader.java:79) ~[formats-gpl-6.4.0.jar:6.4.0]
	at loci.formats.ImageReader.isThisType(ImageReader.java:858) ~[formats-api-6.4.0.jar:6.4.0]
	at loci.formats.ImageReader.getReader(ImageReader.java:189) ~[formats-api-6.4.0.jar:6.4.0]
	at loci.formats.ImageReader.setId(ImageReader.java:840) ~[formats-api-6.4.0.jar:6.4.0]
	at LargeNDPIBug.main(LargeNDPIBug.java:21) ~[main/:na]
```

The issue affects Orbit Image Analysis, but can be worked around by using the native libraries from Hamamatsu to read the files instead. I'm pretty new to the image analysis field so not really sure how to dig into the issue further.

melissalinkert, 2020-04-09:
This looks like an issue with file type detection, as the exception is coming from ```SEQReader``` (when ```NDPIReader``` should be used).  If you modify your test code to use an ```NDPIReader``` instead of ```ImageReader```, do you still see an error?

jfkotw, 2020-04-09:
Looks like that solves the issue. I'd need to do more testing to be certain, but thanks for the hint. It would still be great to fix the file type detection if possible. Let me know if you would still need an example image to do that.

melissalinkert, 2020-04-09:
A test file would still be helpful, if possible.  If needed, I can send you instructions for FTPing files to the OME team.

jfkotw, 2020-04-14:
Hi Melissa, I was going to upload at: https://www.openmicroscopy.org/qa2/qa/upload/
But realised that there is a 2GB file size limit. Ideally the image could live here longer term: https://downloads.openmicroscopy.org/images/Hamamatsu-NDPI/manuel/

Could you post details for sharing by FTP?

melissalinkert, 2020-04-14:
@jfkotw: I have just sent FTP credentials to you via private email (subject: ```Sending large .ndpi files to the OME team```).  If you don't see an email shortly, let me know and we'll work out some other way without posting credentials here publicly.

jfkotw, 2020-04-15:
I didn't get an email, but tweeted @bioformats so that we can share email addresses. Thanks!

melissalinkert, 2020-04-20:
FTP info sent a few minutes ago, so hopefully you're able to connect now. Please let us know if you don't see the email or run into any problems.

melissalinkert, 2020-04-21:
Thanks for uploading the file, @jfkotw.

Using your test code (https://github.com/jfkotw/LargeNDPIBug/commit/e877fe9da2ed878251d729dedd267376ad6fca96), I do see the stack trace initially reported, but it is not fatal.  It's just debug logging to indicate that the ```SEQReader``` can't be used.  ```NDPIReader``` is correctly used in the end, and the file initializes successfully.  Changing the logging level from ```DEBUG``` to ```INFO``` or higher would eliminate the stack traces.  ```showinf -nopix -noflat doubleSizeSlide.ndpi``` with 6.4.0+ can confirm that the correct reader is being used by default.

I don't think there is anything we can fix here, as the file appears to work as expected.

jfkotw, 2020-04-22:
Thanks Melissa for your help here. I verified and you are (of course) exactly right.

I did a quick test and found that the ImageReader might be about 1 second slower than than NDPIReader. I guess that comes down to the fact that the SEQReader is tried first. I'll close the ticket if that is ok?

closed 2020-04-22 by jfkotw

melissalinkert, 2020-04-23:
Yes, ```ImageReader``` is always expected to be a little slower in the ```setId``` call compared to using a particular format reader directly.  ```ImageReader``` has to search through the list of available readers to find the most appropriate one, which can take time especially for TIFF-based file formats (including NDPI).  You can see how this works here:

https://github.com/ome/bioformats/blob/v6.4.0/components/formats-api/src/loci/formats/ImageReader.java#L160