ome/bioformats #3547: Table.save issues when using Bioformat importer and Ext.()

IrrelevantNameTom, 2020-04-20:
Hey everyone,

I wrote a small ImageJ macro to extract metadata from .czi files contained into a folder. The whole procedure is rather straightforward but for some reasons it seems that the built-in function "Table.save()" is not able to overwrites an existing table if any kind of Ext.*() function has been called. 

Error windows "Bio-formats importer: Macro error"
Console "[ERROR] File has length 0 and may be corrupt"

Here is the code to replicate the error (happens the second time when the first file will have been created).

```
dirmain = getDirectory("Choose a Directory ");
list = getFileList(dirmain);

title = "Metadata";
Table.create(title);

for (i=0; i<list.length; i++) {
	ImagePath=dirmain + list[i];
	run("Bio-Formats Macro Extensions");	
	Ext.setId(ImagePath);	
}
	
Table.save(dirmain+"MetaData.csv");
```

Thanks ! : )

dgault, 2020-04-21:
Hi @IrrelevantNameTom , I believe what is happening here is that the csv file is being saved in the imaging directory the first time it is run, then the second time it is included in the filelisting and Bio-Formats is attempting to open it as an image file. Simply saving the csv to a separate directory or excluding it from the generated file list should allow for the macro to continue running without exception.

IrrelevantNameTom, 2020-04-21:
That worked, thanks dgault !

Indeed, it seems that bioformat does not fail to open a .csv file but crashs if trying to apply a Ext;.*()
As I wanted to write the file into the same folder anyway, I managed to shorten the list.length so that bioformat never opens the .csv file.

Code looks like this now.

```
dirmain = getDirectory("Choose a Directory ");
list = getFileList(dirmain);

Pretitle = "MetaData";
title = Pretitle;
Table.create(title);

Dupli=File.exists(dirmain+"\\"+Pretitle+".csv"); 

texiste=0;
while (Dupli==1){
	texiste++;
	title=Pretitle+"_"+texiste;
	Dupli=File.exists(title);
}

for (i=0; i<list.length-texist; i++) {
	ImagePath=dirmain + list[i];
	run("Bio-Formats Macro Extensions");	
	Ext.setId(ImagePath);	
}
	
Table.save(dirmain+Pretitle+".csv");
```

Thanks again !

closed 2020-04-21 by IrrelevantNameTom