ome/bioformats #3560: CZI: Make use of TimeSpan Units

dgault, 2020-05-12:
This issue was first reported in https://forum.image.sc/t/czi-reader-illuminationtype-mapping-and-exposuretime-units/37557/3 and a sample file showing the issue is available in QA-29273

In the sample file provided the ExposureTime is seen as:
```
200.0 s
4000.0 s
```
Currently this is hardcoded in the reader as Seconds, but looking at the metadata of the sample file shows a tag which may be used to provide the correct time units:
`Experiment|AcquisitionBlock|TimeSeriesSetup|Interval|TimeSpan|DefaultUnitFormat #1	ms`

dgault, 2020-07-09:
An additional report of this was received in QA-29404 and a sample file provided in QA-29403
The issue was reproduced using Bio-Formats 6.5.1

With that sample file the exposure times also have been ordered incorrectly (perhaps by size):
From the metadata the correct values are as below (expected units is ms):
```
Information|Image|Channel|ExposureTime #1 = 79999700
Information|Image|Channel|ExposureTime #2 = 1399999700
Information|Image|Channel|ExposureTime #3 = 184999700
Information|Image|Channel|ExposureTime #4 = 100000000
```

Bio-Formats reports these in seconds:
```
<Plane DeltaT=“1.0810000896453857” DeltaTUnit=“s” ExposureTime=“79.999751988515” ExposureTimeUnit=“s” PositionX=“-21332.75” PositionXUnit=“reference frame” PositionY=“-40943.25” PositionYUnit=“reference frame” PositionZ=“562.488” PositionZUnit=“reference frame” TheC=“0” TheT=“0" TheZ=“0”/>
     <Plane DeltaT=“1.937999963760376" DeltaTUnit=“s” ExposureTime=“100.0" ExposureTimeUnit=“s” PositionX=“-21332.75" PositionXUnit=“reference frame” PositionY=“-40943.25" PositionYUnit=“reference frame” PositionZ=“562.488" PositionZUnit=“reference frame” TheC=“1" TheT=“0” TheZ=“0"/>
     <Plane DeltaT=“4.105999946594238” DeltaTUnit=“s” ExposureTime=“1399.9997519885” ExposureTimeUnit=“s” PositionX=“-21332.75” PositionXUnit=“reference frame” PositionY=“-40943.25” PositionYUnit=“reference frame” PositionZ=“562.488” PositionZUnit=“reference frame” TheC=“2” TheT=“0" TheZ=“0”/>
     <Plane DeltaT=“5.240999937057495" DeltaTUnit=“s” ExposureTime=“184.99975198852" ExposureTimeUnit=“s” PositionX=“-21332.75" PositionXUnit=“reference frame” PositionY=“-40943.25" PositionYUnit=“reference frame” PositionZ=“562.488" PositionZUnit=“reference frame” TheC=“3" TheT=“0” TheZ=“0"/>
```

And OMERO displays as:
```
1min 20 sec
1min 40 sec 
23min 20 sec
3min 5sec
```

dgault, 2021-03-01:
And additional report of this issue has been received on https://github.com/ome/openmicroscopy/issues/6266

Migrating the original comment from @CFGrote which is below:

Summary
We experience problems with how exposure time is reported in in the Acquisition tab under Image. Reported values are off by factor 1000.

Details:
The exposure times for 3 channels are set to 25, 500, and 1000 ms, respectively and these values are correctly recorded in the czi file as reported by Zen:

[image]

In omero, under Aqquisition/Image I get these values:
Bright: 25s (=25000 ms)
RFP: 16min40s (=1000000 ms)
EGFP: 8min20s (=500000 ms)

[image]

NOTE: The wavelengths are wrong, too, but that's a problem on our side.

"Original metadata" has two entries per channel that match "ExposureTime":
"Experiment|AcquisitionBlock|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime":
[image]

These values are correct.

The 2nd entry is
"Information|Image|Channel|ExposureTime":
[image]

These values are off by 3 powers of 10 and it looks to me that this is what omero bases it's report in the Acquisition tab on.

We're in touch with people at Zeiss to clarify the meaning of the different exposure time entries in the czi metadata.

Thanks for any suggestions.

Omero Version 5.6.2

Example czi to reproduce this issue:
https://demo.openmicroscopy.org/webclient/?show=image-165834

jaybee2, 2023-03-01:
Hi, I am also experiencing this issue. I took a fluorescence microscopy timeseries with 4 channels. When viewing the output .czi file OME Metadata in FIJI it shows an exposure time of seconds, which I think is off by a factor of 1000 (similar to the above post). 
[image]