ome/bioformats #3565: ND2 time lapse time stamps are not read correctly

manerotoni, 2020-05-18:
Dear OME team, 
I have an issue with reading ND2 files from NIS-elements. In particular with the time-stamps. I attach a file. 
The expected time is 
0.278270
10.017
20.033

but bioformat version 6.5.1-SNAPSHOT reads
<Plane DeltaT="0.07654202796406791" DeltaTUnit="s" >
<Plane DeltaT="-3.7877675682026275E-121" DeltaTUnit="s" >
<Plane DeltaT="-1.3375473337470363E250" DeltaTUnit="s" >

Thanks

Antonio
[example_metadata001.zip](https://github.com/ome/bioformats/files/4645173/example_metadata001.zip)

the time stamp that are read by Bioformats are not correct.

manerotoni, 2020-05-18:
This issue may be similar to 
https://github.com/ome/bioformats/issues/3164

dgault, 2020-05-19:
Hi @manerotoni, thank you opening the issue and providing a sample file. Does this affect all your ND2 files? Have you made any recent updates to NIS-elements or made any other updates that have led to this?

manerotoni, 2020-05-19:
Unfortunately, I do not know if the issue is new or not as we just started to regularly do time-lapses now. The version of the NIS-elements we are using is 5.21.01 b1483 which is a rather recent version.

AleixBP, 2020-06-08:
Hello!
I have the same issue. These are the time stamps I get:
timestamp #01	8.45596649986333E-310
timestamp #02	8.90114424711E-311
timestamp #03	4.0052925225049E-310
timestamp #04	1.602203926757746E-309
Did you manage to find the problem? Please let me know if I can help in any way. I can provide test data and any other specification that might be useful.

I would like to take the chance to ask a second question. Is there any possibility I could eventually get the time stamps as computer clock time (including year, month, day, minute, etc.)? Or the microscope just does not output that?

Thank you very much for your consideration.

All the best,

Aleix

dgault, 2020-07-22:
A further report of the same issue can be found on https://forum.image.sc/t/nd2-metadata-and-import-issues/40699/2

dgault, 2020-07-27:
An additional sample file relating to this issue is available in QA-29438

quantixed, 2021-06-11:
I am bumping this issue since I see the same behaviour:

```
<Plane DeltaT="0.0" DeltaTUnit="s" ExposureTime="0.08" ExposureTimeUnit="s" PositionX="-7791.400000000001" PositionXUnit="reference frame" PositionY="-4232.2" PositionYUnit="reference frame" PositionZ="3180.98" PositionZUnit="reference frame" TheC="0" TheT="98" TheZ="0"/>
<Plane DeltaT="0.0" DeltaTUnit="s" ExposureTime="0.08" ExposureTimeUnit="s" PositionX="-7791.400000000001" PositionXUnit="reference frame" PositionY="-4232.2" PositionYUnit="reference frame" PositionZ="3180.98" PositionZUnit="reference frame" TheC="0" TheT="99" TheZ="0"/>
<Plane DeltaT="2.584679942E-314" DeltaTUnit="s" ExposureTime="0.08" ExposureTimeUnit="s" PositionX="-7791.400000000001" PositionXUnit="reference frame" PositionY="-4232.2" PositionYUnit="reference frame" PositionZ="3180.98" PositionZUnit="reference frame" TheC="0" TheT="100" TheZ="0"/>
<Plane DeltaT="1.5E-323" DeltaTUnit="s" ExposureTime="0.08" ExposureTimeUnit="s" PositionX="-7791.400000000001" PositionXUnit="reference frame" PositionY="-4232.2" PositionYUnit="reference frame" PositionZ="3180.98" PositionZUnit="reference frame" TheC="0" TheT="101" TheZ="0"/>
<Plane DeltaT="3.589901569625176E157" DeltaTUnit="s" ExposureTime="0.08" ExposureTimeUnit="s" PositionX="-7791.400000000001" PositionXUnit="reference frame" PositionY="-4232.2" PositionYUnit="reference frame" PositionZ="3181.0" PositionZUnit="reference frame" TheC="0" TheT="102" TheZ="0"/>
<Plane DeltaT="3.171126472878828E177" DeltaTUnit="s" ExposureTime="0.08" ExposureTimeUnit="s" PositionX="-7791.400000000001" PositionXUnit="reference frame" PositionY="-4232.2" PositionYUnit="reference frame" PositionZ="3180.98" PositionZUnit="reference frame" TheC="0" TheT="103" TheZ="0"/>
<Plane DeltaT="4.824065448409745E225" DeltaTUnit="s" ExposureTime="0.08" ExposureTimeUnit="s" PositionX="-7791.400000000001" PositionXUnit="reference frame" PositionY="-4232.2" PositionYUnit="reference frame" PositionZ="3181.0" PositionZUnit="reference frame" TheC="0" TheT="104" TheZ="0"/>
<Plane DeltaT="7.04335894E-316" DeltaTUnit="s" ExposureTime="0.08" ExposureTimeUnit="s" PositionX="-7791.400000000001" PositionXUnit="reference frame" PositionY="-4232.2" PositionYUnit="reference frame" PositionZ="3181.0" PositionZUnit="reference frame" TheC="0" TheT="105" TheZ="0"/>
<Plane DeltaT="0.0" DeltaTUnit="s" ExposureTime="0.08" ExposureTimeUnit="s" PositionX="-7791.400000000001" PositionXUnit="reference frame" PositionY="-4232.2" PositionYUnit="reference frame" PositionZ="3180.98" PositionZUnit="reference frame" TheC="0" TheT="106" TheZ="0"/>
```
The timestamps for this file can be read correctly by NIS Elements. Similar region of movie is shown below

```
01:20.8	93	-7791.4	-4232.2	3180.98	3280.78	99.8	6868	-14.96	80	0
01:21.8	94	-7791.4	-4232.2	3180.98	3280.78	99.8	6868	-14.96	80	0
01:22.8	95	-7791.4	-4232.2	3180.98	3280.78	99.8	6868	-15.01	80	0
01:23.8	96	-7791.4	-4232.2	3180.98	3280.78	99.8	6868	-15.01	80	0
01:24.8	97	-7791.4	-4232.2	3180.98	3280.78	99.8	6868	-15.01	80	0
01:25.8	98	-7791.4	-4232.2	3180.98	3280.78	99.8	6868	-15.01	80	0
```
It is possible to export them as a spreadsheet from there, but it would be great if BioFormats could read this metadata correctly.

manerotoni, 2021-06-11:
Dear @quantixed,
I try to make Nikon aware about this problem since several months. My impression is that it is a problem on their side or that Nikon should get in contact with OME and work on a fix. According to the forum issue https://forum.image.sc/t/nd2-video-frame-time-stamp-corruption-in-imagej-metadata/38012/31 it seems that since 2020 the metadata is corrupted. 

May be you can also write to instruments-support.eu@nikon.com to make them aware that I am not the only person having an issue with the time-stamp.

Antonio

quantixed, 2021-06-11:
Hi @manerotoni, will do.

quantixed, 2021-06-22:
@manerotoni Nikon got back to me - some good news:

> One time stamp issue is fixed and will be released in next Bio-Formats 6.7.0 release.
https://github.com/ome/bioformats/pull/3692
The other is pending:
https://github.com/ome/bioformats/issues/3502

manerotoni, 2021-06-23:
These are great news. Thanks for taking care of it. I am sure also the other issue will be solved eventually

melissalinkert, 2023-08-11:
Retested the example file in https://github.com/ome/bioformats/issues/3565#issue-620316960 with 7.0.0. The timestamps now match what is shown in NIS Elements Viewer.

closed 2023-08-11 by melissalinkert