ome/bioformats #3570: Add Ventana bif 'overlap.direction.equals("LEFT")'

Tato14, 2020-06-11:
It is already know that some Ventana bif file fail to open because of the "LEFT" `TileJointInfo` field.

I found this [commit](https://github.com/openslide/openslide/pull/246/files) not merged on `openslide`  `master` that I hope that could shed some light on how to solve the issue.

I seems that the code does not change too much but I'm not sure if this apply in Bio-formats.

sbesson, 2020-06-11:
Thanks for opening an issue for this restriction which has been reported twice in the image.sc forum in https://forum.image.sc/t/release-of-bio-formats-6-2-1/29132/2 and https://forum.image.sc/t/problems-with-ventana-bif-images/38863.

First, do you have some Ventanta BIF files representative of the issue that could be shared with the OME team, ideally sample files that could be [shared publicly](https://downloads.openmicroscopy.org/images/Ventana/) under a CC-BY 4.0 license?

In terms of the code, the  code portion in the Ventana reader dealing with the overlap direction is

https://github.com/ome/bioformats/blob/ed992ebc2cb07d04fcb49485540af0babd1baa05/components/formats-gpl/src/loci/formats/in/VentanaReader.java#L566-L586

Supporting `LEFT` field would mean adding some code handling the scenario `overlap.direction.equals("LEFT")`. Similarly to https://github.com/openslide/openslide/pull/246#issuecomment-407196071, do you have a feeling for the differences between `LEFT` and `RIGHT`?

Tato14, 2020-06-12:
Apologize for cross-posting.

I am speaking with our institution in order to see how we can release an image of this kind in the following days.

In the meantime I've built Bio-Formats from source using 6.5.1-SNAPSHOT from github and  `mvn` command. I changed the code you referenced to:

```
for (Overlap overlap : area.overlaps) {
        if (overlap.confidence < 98) {
          continue;
        }
        if (overlap.direction.equals("RIGHT")) {
          rightSum += overlap.x;
          rightCount++;
          if (overlap.y > 0) {
            columnYAdjust.put(getTileColumn(
              overlap.a, area.tileRows, area.tileColumns), overlap.y);
          }
        }
        else if (overlap.direction.equals("UP")) {
          upSum += overlap.y;
          upCount++;
        }
        else if (overlap.direction.equals("LEFT")) {
          rightSum += overlap.x;
          rightCount++;
          if (overlap.y > 0) {
            columnYAdjust.put(getTileColumn(
              overlap.a, area.tileRows, area.tileColumns), overlap.y);
          }
        }
        else {
          throw new FormatException(
            "Unsupported overlap direction: " + overlap.direction);
        }
      }
```
In order to see what error it generates:

Replacing `bioformats_package.jar` with the new jar file in the bftools folder and running:
`./showinf -nopix  file.bif`

I got:
```
Checking file format [Ventana .bif]
Initializing reader
VentanaReader initializing file.bif
Reading IFDs
Populating metadata
Populating OME metadata
Exception in thread "main" java.lang.IllegalArgumentException: -2147319808 must be neither null nor strictly negative.
	at ome.xml.model.primitives.NonNegativeInteger.<init>(NonNegativeInteger.java:48)
	at ome.xml.model.primitives.PositiveInteger.<init>(PositiveInteger.java:46)
	at loci.formats.MetadataTools.populatePixelsOnly(MetadataTools.java:311)
	at loci.formats.MetadataTools.populateMetadata(MetadataTools.java:262)
	at loci.formats.MetadataTools.populatePixels(MetadataTools.java:151)
	at loci.formats.MetadataTools.populatePixels(MetadataTools.java:116)
	at loci.formats.in.BaseTiffReader.initMetadataStore(BaseTiffReader.java:425)
	at loci.formats.in.VentanaReader.initMetadataStore(VentanaReader.java:743)
	at loci.formats.in.BaseTiffReader.initMetadata(BaseTiffReader.java:98)
	at loci.formats.in.BaseTiffReader.initFile(BaseTiffReader.java:586)
	at loci.formats.FormatReader.setId(FormatReader.java:1389)
	at loci.formats.ImageReader.setId(ImageReader.java:843)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1035)
	at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121)
```

I see that the negative value it prints in the first error line its file dependent. Guessing that this have something to do with image dimension, in this specific case the WSI dimensions are:

```
[user@oslo bftools]$ identify file.bif 
file.bif[0] TIFF64 1251x3685 1251x3685+0+0 8-bit DirectClass 4.0946GB 0.050u 0:00.140
file.bif[1] TIFF64 1251x3685 1251x3685+0+0 8-bit Grayscale DirectClass 4.0946GB 0.050u 0:00.130
file.bif[2] TIFF64 90112x163840 90112x163840+0+0 8-bit DirectClass 4.0946GB 0.030u 0:00.080
file.bif[3] TIFF64 45056x81920 45056x81920+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.020
file.bif[4] TIFF64 22528x40960 22528x40960+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.020
file.bif[5] TIFF64 11264x20480 11264x20480+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.010
file.bif[6] TIFF64 5632x10240 5632x10240+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.010
file.bif[7] TIFF64 2816x5120 2816x5120+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.000
file.bif[8] TIFF64 1408x2560 1408x2560+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.000
file.bif[9] TIFF64 704x1280 704x1280+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.000
file.bif[10] TIFF64 352x640 352x640+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.000
file.bif[11] TIFF64 176x320 176x320+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.000
file.bif[12] TIFF64 88x160 88x160+0+0 8-bit DirectClass 4.0946GB 0.000u 0:00.000
```

melissalinkert, 2020-06-12:
@Tato14 : Does removing the ```overlap.y > 0``` condition (or reversing it to ```overlap.y <= 0```) for ```LEFT``` overlaps work?  ```columnYAdjust``` is used to record how columns of tiles should be vertically shifted relative to each other; I would expect a ```LEFT``` overlap to vertically shift in the opposite direction of a ```RIGHT``` overlap.

https://github.com/libvips/libvips/issues/1189#issuecomment-449628383 has a relevant .bif file, which is in ```inbox/gh3570/``` for our internal testing.

Tato14, 2020-06-15:
@melissalinkert 
It seems that `overlap.y <= 0` work nicely in my images. Here is the output after building from source with the change:
```
./showinf -nopix file.bif
Checking file format [Ventana .bif]
Initializing reader
VentanaReader initializing file.bif
Reading IFDs
Populating metadata
Populating OME metadata
Initialization took 1.256s

Reading core metadata
filename = file.bif
Series count = 13
Series #0 :
	Image count = 1
	RGB = true (3) 
	Interleaved = false
	Indexed = false (true color)
	Width = 85536
	Height = 164864
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 65 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (uncertain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #1 :
	Image count = 1
	RGB = true (3) 
	Interleaved = false
	Indexed = false (true color)
	Width = 42768
	Height = 82432
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 65 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (uncertain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = true
	Thumbnail series = true
	-----
	Plane #0 <=> Z 0, C 0, T 0

Series #2 :
	Image count = 1
	RGB = true (3) 
	Interleaved = false
	Indexed = false (true color)
	Width = 21384
	Height = 41216
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Tile size = 1024 x 1024
	Thumbnail size = 65 x 128
	Endianness = intel (little)
[... 235 lines cut ...]

```
identify file.bif
file.bif[0] TIFF64 1251x3685 1251x3685+0+0 8-bit DirectClass 4.8565GB 0.060u 0:00.120
file.bif[1] TIFF64 1251x3685 1251x3685+0+0 8-bit Grayscale DirectClass 4.8565GB 0.060u 0:00.109
file.bif[2] TIFF64 86016x164864 86016x164864+0+0 8-bit DirectClass 4.8565GB 0.030u 0:00.039
file.bif[3] TIFF64 43008x82432 43008x82432+0+0 8-bit DirectClass 4.8565GB 0.010u 0:00.009
file.bif[4] TIFF64 21504x41216 21504x41216+0+0 8-bit DirectClass 4.8565GB 0.010u 0:00.000
file.bif[5] TIFF64 10752x20608 10752x20608+0+0 8-bit DirectClass 4.8565GB 0.000u 0:00.000
file.bif[6] TIFF64 5376x10304 5376x10304+0+0 8-bit DirectClass 4.8565GB 0.000u 0:00.000
file.bif[7] TIFF64 2688x5152 2688x5152+0+0 8-bit DirectClass 4.8565GB 0.000u 0:00.000
file.bif[8] TIFF64 1344x2576 1344x2576+0+0 8-bit DirectClass 4.8565GB 0.000u 0:00.000
file.bif[9] TIFF64 672x1288 672x1288+0+0 8-bit DirectClass 4.8565GB 0.000u 0:00.000
file.bif[10] TIFF64 336x644 336x644+0+0 8-bit DirectClass 4.8565GB 0.000u 0:00.000
file.bif[11] TIFF64 168x322 168x322+0+0 8-bit DirectClass 4.8565GB 0.000u 0:00.000
file.bif[12] TIFF64 84x161 84x161+0+0 8-bit DirectClass 4.8565GB 0.000u 0:00.000
```
In any case, if I open the slide both with `./showinf -series 11 file.bif` 

[image]
(I blurred the patient information)

I am generating an ome.tiff of the file using `bfconvert` in order to check everything looks nice in QuPath. Will come back with the update as soon as the conversion is done

Regarding the image you linked, i do not think this is an example of `LEFT` Roche file. This ones seems to be generated by DP200 scanners and I didn't find anyone publicly available.

sbesson, 2020-06-15:
@Tato14 that's excellent news. I suspect the file Melissa was referring to is the one linked from [this comment](https://github.com/libvips/libvips/issues/1189#issuecomment-449677327) which fails with `loci.formats.FormatException: Unsupported overlap direction: LEFT` for me locally.

Would you be okay to open a Pull Request against this repository with your code changes so that we can confirm it has no impact on the rest of the Ventana files in our curated QA repository?

Tato14, 2020-06-16:
@sbesson PR sent with the code update. Hope it works well!

melissalinkert, 2022-03-30:
Closing since #3573 fixed this and has been included since 6.5.1.

closed 2022-03-30 by melissalinkert

pmshiva, 2022-05-18:
Hi,
   Could anyone please help in rebuilding openslide-python with LEFT attribute added to support Ventana BIF images??

sbesson, 2022-05-18:
@pmshiva `openslide-python` is managed by a separate group of people, have you tried opening an issue against https://github.com/openslide/openslide-python/issues?