ome/bioformats #3574: Feature request: Disable thumbnails in BF plugin

dgault, 2020-06-18:
This is a request which was received via https://forum.image.sc/t/bio-formats-xml-hdf5-reader-performance/30458

When using the Bio-Formats plugin in FIJI it would be useful to provide the option to disable thumbnail generation in order to improve performance at the Series select window.

dgault, 2020-11-25:
Another request for this feature has been received in forum thread: https://forum.image.sc/t/speeding-up-metamorph-file-reading-with-bioformats/42532/17

From the thread:

> For metamorph file I have a time lapse of 179 time points and 128 positions, each time point is a separate stk file with 3 Z-slices a 28 MB. All in all over 22000 files.
> Just the opening of the series choice menu takes between 1 and 2 hour.

manerotoni, 2021-10-18:
Hello, 
are there any news on implementing a nopix opening in bioformat? Generally improving the reading in of multi-file data?

Currently 18720 images (multi-position+time-lapse in total 294 GB) takes several hours until one gets the first response of bioformat in Fiji. I basically gave up. Currently I am reading each position as image-sequence and this is truly not the right choice. 

Using the command line tool, reading of the information with the non-pix option is done within several minutes (which is still quite long).

I guess this is also a problem for OMERO as it can't load this kind of data in an efficient way.