ome/bioformats #3589: ND2: IllegalArgumentException: 0 must be non-null and strictly positive

dgault, 2020-07-20:
This issue was raised on the thread https://forum.image.sc/t/exception-issue-when-opening-nd2-files-with-bio-formats/40462
A sample file has been provided in QA-29421 and the exception is reproduced with Bio-Formats 6.5.1

From debugging the issue is caused by the height (which should be 1024) getting reset to 0 at https://github.com/ome/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/NativeND2Reader.java#L1246

This is due to availableBytes being 0 which is the bigger issue and is set at https://github.com/ome/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/NativeND2Reader.java#L1124

The offsets and other metadata values largely look to be correct and sensible. I have not yet been able to determined if the first offset is indeed incorrect or if our handling of it is to blame.

dgault, 2020-09-14:
An additional report of this issue has been received and a new sample file showing the exception is available in QA-29536

Roos2022, 2022-06-20:
Hi. I encountered the same issue when uploading ND2 vials, although not with all files. Is it possible to use the files that you send to open my files? If yes, could you tell me how? Thank you!

Jimmy3times, 2022-10-20:
Hi,

I have the same problem. It seem size of the file is an issue - files above 2GB will not open, Bio-Formats import function does not help either. 
Any idea how go around this problem (full disclosure - coding is not my thing, I'm just trying to analyze large images).?

Jimmy3times, 2022-10-20:
Error message :
(Fiji Is Just) ImageJ 2.9.0/1.53t; Java 1.8.0_66 [64-bit]; Windows 10 10.0; 3036MB of 98127MB (3%)
 
java.lang.IllegalArgumentException: Array size too large: 25745 x 42803 x 1 x 2
	at loci.common.DataTools.safeMultiply32(DataTools.java:1286)
	at loci.common.DataTools.allocate(DataTools.java:1259)
	at loci.formats.TileStitcher.openBytes(TileStitcher.java:119)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
	at loci.plugins.util.ImageProcessorReader.openProcessors(ImageProcessorReader.java:186)
	at loci.plugins.in.ImagePlusReader.readProcessors(ImagePlusReader.java:422)
	at loci.plugins.in.ImagePlusReader.readPlanes(ImagePlusReader.java:387)
	at loci.plugins.in.ImagePlusReader.readImage(ImagePlusReader.java:282)
	at loci.plugins.in.ImagePlusReader.readImages(ImagePlusReader.java:243)
	at loci.plugins.in.ImagePlusReader.readImages(ImagePlusReader.java:221)
	at loci.plugins.in.ImagePlusReader.openImagePlus(ImagePlusReader.java:116)
	at loci.plugins.in.Importer.readPixels(Importer.java:149)
	at loci.plugins.in.Importer.run(Importer.java:86)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:237)
	at ij.IJ.runPlugIn(IJ.java:203)
	at ij.IJ.runPlugIn(IJ.java:192)
	at HandleExtraFileTypes.openImage(HandleExtraFileTypes.java:524)
	at HandleExtraFileTypes.run(HandleExtraFileTypes.java:97)
	at ij.IJ.runUserPlugIn(IJ.java:237)
	at ij.IJ.runPlugIn(IJ.java:203)
	at ij.IJ.runPlugIn(IJ.java:192)
	at ij.io.Opener.openWithHandleExtraFileTypes(Opener.java:555)
	at ij.io.Opener.openUsingHandleExtraFileTypes(Opener.java:407)
	at ij.io.Opener.openImage(Opener.java:386)
	at ij.io.Opener.openImage(Opener.java:241)
	at ij.io.Opener.open(Opener.java:104)
	at ij.io.Opener.openAndAddToRecent(Opener.java:310)
	at ij.plugin.DragAndDrop.openFile(DragAndDrop.java:194)
	at ij.plugin.DragAndDrop.run(DragAndDrop.java:160)
	at java.lang.Thread.run(Thread.java:745)

melissalinkert, 2022-10-21:
@Jimmy3times: is the `Stitch tiles` option checked in the `Bio-Formats Import Options` window? If so, does unchecking that option allow the file to be opened?

Jimmy3times, 2022-10-21:
@melissalinkert - was checked, I unchecked it, and it didn't change much - same error :/

Jimmy3times, 2022-10-26:
Anybody has any idea? @dgault - you seemed to investigate this issue a while back, do you know if some fix is available?

dgault, 2022-11-15:
Sorry, must have missed this comment while I was away, unfortunately we don't have an available fix for this issue at the moment.