ome/bioformats #3591: "SignificantBits" field is ignored in Zeiss CZI files

jamescooper-gd, 2020-07-22:
Auxiliary images (label etc) in Zeiss CZI files often appear fully black.  Upon inspection it seems that they have the `SignificantBits` attribute set to a value smaller than the width of the `Type`.

E.g.
`<Pixels BigEndian="false" DimensionOrder="XYCZT" ID="Pixels:16" Interleaved="true" PhysicalSizeX="0.22070127831183586" PhysicalSizeXUnit="µm" PhysicalSizeY="0.22070127831183586" PhysicalSizeYUnit="µm" SignificantBits="12" SizeC="3" SizeT="1" SizeX="722" SizeY="531" SizeZ="1" Type="uint16">`

To address this in my own code I inspect the `IMetadata.getPixelsSignificantBits(int)` value, then adjust the buffer values after opening the image:

```java
private BufferedImage openImage(BufferedImageReader r, int ch, int x, int y, int w, int h) throws FormatException, IOException {
    BufferedImage img = r.openImage(ch, x, y, w, h);
    if (pixelsSignificantBits != -1) { // Bioformats doesn't take into account the OME "significant bits" field
        int pixelType = r.getPixelType();
        int bpp = 8 * FormatTools.getBytesPerPixel(pixelType);
        int shifts = bpp - pixelsSignificantBits;
        boolean fp = pixelType == FormatTools.FLOAT || pixelType == FormatTools.DOUBLE;
        if (shifts > 0 && !fp) {
            apply(img.getRaster().getDataBuffer(), i -> i << shifts);
        }
    }
    return img;
}

private static final void apply(DataBuffer buffer, IntUnaryOperator function) {
    if (buffer instanceof DataBufferByte) {
        applyBytes(((DataBufferByte) buffer).getData(), function);
    } else if (buffer instanceof DataBufferShort) {
        applyShorts(((DataBufferShort) buffer).getData(), function);
    } else if (buffer instanceof DataBufferUShort) {
        applyShorts(((DataBufferUShort) buffer).getData(), function);
    } else if (buffer instanceof DataBufferInt) {
        applyInts(((DataBufferInt) buffer).getData(), function);
    } else if (buffer instanceof SignedByteBuffer) {
        applyBytes(((SignedByteBuffer) buffer).getData(), function);
    } else if (buffer instanceof SignedShortBuffer) {
        applyShorts(((SignedShortBuffer) buffer).getData(), function);
    } else if (buffer instanceof UnsignedIntBuffer) {
        applyInts(((UnsignedIntBuffer) buffer).getData(), function);
    } else {
        throw new UnsupportedOperationException("Unsupported buffer type: " + buffer.getClass());
    }
}

private static final void applyBytes(byte[] vv, IntUnaryOperator function) {
    for (int i = 0; i < vv.length; i++) {
        vv[i] = (byte) function.applyAsInt(0xff & vv[i]);
    }
}

private static final void applyShorts(short[] vv, IntUnaryOperator function) {
    for (int i = 0; i < vv.length; i++) {
        vv[i] = (short) function.applyAsInt(0xffff & vv[i]);
    }
}

private static final void applyInts(int[] vv, IntUnaryOperator function) {
    for (int i = 0; i < vv.length; i++) {
        vv[i] = function.applyAsInt(vv[i]);
    }
}
```

I'm not sure whether the fix should be in all readers, or just Zeiss.  The other file formats I have access to don't use the `SignificantBits` attribute, but I only have access to a limited set of real images.

dgault, 2020-07-23:
Hi @jamescooper-gd, thank you for opening the issue. The Bio-Formats library itself doesn't modify or scale any pixel values returned. So for something such SignifigantBits it will very much be upto the API consumer to handle as they see best. 

In this case the way your interpreting it looks to be correct. If it is of use there are some helper methods in [ImageTools](https://github.com/ome/bioformats/blob/develop/components/formats-bsd/src/loci/formats/ImageTools.java ) which may be of use, such as [make24Bits](https://github.com/ome/bioformats/blob/develop/components/formats-bsd/src/loci/formats/ImageTools.java#L107) which could 0 and SignificantBits as your min and max or for [autoscaling](https://github.com/ome/bioformats/blob/develop/components/formats-bsd/src/loci/formats/ImageTools.java#L468.)

jamescooper-gd, 2020-07-23:
I can understand that view.  I filed the issue because Fiji and QuPath both suffer from the same issue as our software did.  While I think it's correct that `openBytes()` doesn't scale anything I _don't_ think it's correct that `openImage()` doesn't.  It seems that the lib tries hard to ensure that the samples are untouched, but that the `ColorModel` is such that the image appears correct.  My expectation of the library is that this would hold true for the significant bits attribute too.

I understand not wanting to change the default, but I think it's worthwhile considering an `AWTImageTools.openImage()` method that allows the max bits/pixel to be defined. Currently if I want to do it via the `ColorModel`, or via changing the raw buffer as its created, then I'd have to duplicate almost 100 lines of code from `AWTImageTools.openImage(byte[] buf, IFormatReader r, int w, int h, boolean normal)`, since this method already handles all the heavy lifting of handling the various heterogeneous input.

To avoid copying the entire method, the easiest workaround for me is to adjust the raw buffer values after the fact, hence my hack.

What are your thoughts?

dgault, 2020-07-27:
Would you not be able to achieve the same result by using `AWTImageTools.openImage` followed by `AWTImageTools.autoscale(BufferedImage img, int min, int max)` with max set to the desired range?

jamescooper-gd, 2020-07-28:
Yes, that absolutely work, thanks for the alternative.  I was focused the changes I would need in order to get the desired values in a single pass rather than in a post-processing step.

I would still prefer that the library produce an image according to all of the metadata rather than most of it, but understand reticence when it comes to library changes.  If it's insufficient justification that other software tools that use bio-formats also produce erroneous images because this field is ignored, then please close this issue (I've only looked at Fiji and QuPath).

Thanks for the comments.