ome/bioformats #3594: nd2 Multipoint is read as timeseries

gatoniel, 2020-07-24:
Hi,

I created a nd2 file by using NIS Elements Import Feature "Create ND File from File Sequence..." (NIS Elements 4.50.00 Build 1108e). In NIS Elements the file is a multipoint file with 14 Positions. Each Position has three channels. When I use bfconvert on the file the resulting ome.tiff file has only one series with 42 Pages. The bio-formats Importer in Fiji gives me an image with 3 channels and 14 timepoints. I was expecting an image with 3 channels, but 14 series. Is this the intended behaviour?

I uploaded the file via https://www.openmicroscopy.org/qa2/qa/upload/

dgault, 2020-07-27:
Hi @gatoniel, thank you for reporting the issue and uploading the sample file (QA-29438). This isnt the intended behaviour and does look to be a bug. I was able to reproduce what you are seeing with the latest Bio-Formats 6.5.1. It looks like this may be a similar problem to other issues which have previously been reported by users with ND2 positions being misread: https://github.com/ome/bioformats/issues/3502

melissalinkert, 2023-08-11:
Retested this file with 7.0.0; it is now detected as 14 series each with 3 channels, matching what is shown in NIS Elements Viewer. https://github.com/openmicroscopy/data_repo_config/pull/589 adds the file to the test repo to prevent future regressions.

closed 2023-08-11 by melissalinkert