ome/bioformats #3598: SISReader: Image name and channel name includes trailing null byte

skalteis, 2020-07-29:
Hi,

I recently was using QuPath 0.2.1 which uses Bioformats 6.5.1 with some (SIS-)TIFF files created by Olympus cell^F and found that the image name and channel name as provided by the Bioformats reader include the trailing null byte at the end of the string.

I looked up the code in RandomAccessInputStream.java and found that readCString essentially calls ReadString which in my understanding reads through the null byte and returns it along with the string.

I added a call to trim() to the readCString function call in lines 174 and 239 in SISreader.java and this fixes the problem for me.

The image file I used to test this can be found at
https://box.med.uni-heidelberg.de/s/qjr7vvn7P9akLMx
Password for access: QuPathrocks1! (please include the exclamation mark)

The associated QuPath issue is: https://github.com/qupath/qupath/issues/573

Thank you for your work & best regards
Simon

dgault, 2020-07-31:
Thanks @skalteis for opening the issue and providing a test file. I will try your fix against the sample files that we have. If it also works across our data repo then hopefully we can get a fix included for the next Bio-Formats release.

dgault, 2020-08-05:
Thanks again @skalteis for the feedback. Your recommended changes will be included in the next release of Bio-Formats 6.6.0

PR #3602 (2020-08-05): SIS: Trim imageName and channelName
This PR is implementing the recommended changes from https://github.com/ome/bioformats/issues/3598
Opening to test the impact on the data repo

Fixes #3598
closed 2020-08-05 by dgault