ome/bioformats #3607: nii vs nii.gz

dominikl, 2020-08-14:
Importing  unzipped `nii` files works fine, whereas the same files in their original `nii.gz` form take so long that it's practically not possible to import. See https://forum.image.sc/t/importing-large-neuroimaging-data-into-omero/40114 
I'm capturing the problem here, because I'm not sure yet which component is actually causing this.

sbesson, 2020-08-14:
A few additional comments here:

- support for gzipped compressed NIfTI data is relatively recent (it was added via an external contribution in https://github.com/ome/bioformats/pull/2508) and has primarily been tested with the sample NIH datasets (https://nifti.nimh.nih.gov/nifti-1/data). Thus I am not completely surprised by the performance degradation for large volumes as we hold no representative sample of real-world large `.nii.gz`
- the reader logic uses the filename to determine which handle should be used to stream the file which probably explain why the trick suggested in https://github.com/ome/bioformats/issues/3607#issuecomment-671648632 did not work
- if conversion is an option and the NifTi format is not a requirement else where in the pipeline, it might be worth testing the conversion into e.g. zlib-compressed tiled OME-TIFF using either the command-line tools and/or the [bioformats2raw/raw2ometiff pipeline](https://forum.image.sc/t/converting-whole-slide-images-to-ome-tiff-a-new-workflow/32110) as an alternative
- to fix the issue at the reader level, some profiling might help identifying which calls are causing the bottleneck although I am pretty  likely this happens during compression/decompression. Does anyone have experience with how this problem is solved in reference NifTi libraries or other software? Is there some expected benchmark metric for the random access to a file under `.nii` or `.nii.gz` format?

dgault, 2020-08-19:
Was running some quick profiling on this. Opening a small .nii.gz resulted in a very long read time, caused by a seek call for each openBytes:

[image]
Looking at the code I retested after the removing the below line which looks unnecessary:
https://github.com/ome/bioformats/pull/2508/files#diff-1696608d5491b8873c77d696b4f4cc4aR150

Without that line the same file then produced the below times and still loaded as expected:

[image]

Unfortunately that line looks to have been in place to fix test failures and as suggested in https://github.com/ome/bioformats/pull/2508#issuecomment-237065726 it is most likely that there is an underlying issue in GZipHandle which needs addressed. But it does appear that it is the workaround that is resulting in the performance issue

melissalinkert, 2023-06-16:
With #3891 merged and released, is there further work to do here or can this issue be closed?

dgault, 2023-06-19:
Looks like this can now be closed

closed 2023-06-19 by dgault