ome/bioformats #3612: Nanostring GeoMx tiff import incorrectly

olatarkowska, 2020-08-25:
Nanostring GeoMx

1) the image can be imported but the pyramid levels are split as individual files. Could this be fixed or coudl you please let me know about any intermidiet solution?

2) the microscope also outputs the channels as speared files. Can we merge the channels if pyramid is broken?

olatarkowska, 2020-08-25:
Data was shared with @sbesson via globus, let me know if the endpoint needs more people

sbesson, 2020-08-26:
Thanks for sharing the data, I was able to download it locally and am now copying it to a location accessible to the wider OME team.

Short summary of the source data:
- it is composed of 5 files, 4 8GB TIFF file (`NS_DSP_00MR_raw_Cy3.tiff`, `NS_DSP_00MR_raw_Cy5.tiff`, `NS_DSP_00MR_raw_FITC.tiff`, `NS_DSP_00MR_raw_TR.tiff`) and a pattern file `NS_DSP_00MR.pattern` which is combinining these together
- each individual TIFF file containing multiple IFDs with tiled uncompressed images and images sizes are halved by two

 ```
$ tiffinfo NS_DSP_00MR_raw_Cy3.tiff 
TIFF Directory at offset 0x180000010 (6442450960)
  Image Width: 49152 Image Length: 65536
  Tile Width: 256 Tile Length: 256
  Resolution: 25145, 25193 pixels/cm
  Bits/Sample: 16
  Compression Scheme: None
  Photometric Interpretation: min-is-black
  FillOrder: msb-to-lsb
  Samples/Pixel: 1
  Planar Configuration: single image plane
TIFF Directory at offset 0x1e00c014c (8053850444)
  Image Width: 24576 Image Length: 32768
  Tile Width: 256 Tile Length: 256
  Resolution: 12572, 12596 pixels/cm
  Bits/Sample: 16
  Compression Scheme: None
  Photometric Interpretation: min-is-black
  FillOrder: msb-to-lsb
  Samples/Pixel: 1
  Planar Configuration: single image plane
TIFF Directory at offset 0x1f80f0274 (8456700532)
  Image Width: 12288 Image Length: 16384
  Tile Width: 256 Tile Length: 256
  Resolution: 6286, 6298 pixels/cm
  Bits/Sample: 16
  Compression Scheme: None
  Photometric Interpretation: min-is-black
  FillOrder: msb-to-lsb
  Samples/Pixel: 1
  Planar Configuration: single image plane
TIFF Directory at offset 0x1fe0fc39c (8557413276)
  Image Width: 6144 Image Length: 8192
  Tile Width: 256 Tile Length: 256
  Resolution: 3143, 3149 pixels/cm
  Bits/Sample: 16
  Compression Scheme: None
  Photometric Interpretation: min-is-black
  FillOrder: msb-to-lsb
  Samples/Pixel: 1
  Planar Configuration: single image plane
TIFF Directory at offset 0x1ff8ff4c4 (8582591684)
  Image Width: 3072 Image Length: 4096
  Tile Width: 256 Tile Length: 256
  Resolution: 1571, 1574 pixels/cm
  Bits/Sample: 16
  Compression Scheme: None
  Photometric Interpretation: min-is-black
  FillOrder: msb-to-lsb
  Samples/Pixel: 1
  Planar Configuration: single image plane
TIFF Directory at offset 0x1fff001ec (8588886508)
  Image Width: 1536 Image Length: 2048
  Tile Width: 256 Tile Length: 256
  Resolution: 785, 787 pixels/cm
  Bits/Sample: 16
  Compression Scheme: None
  Photometric Interpretation: min-is-black
  FillOrder: msb-to-lsb
  Samples/Pixel: 1
  Planar Configuration: single image plane
TIFF Directory at offset 0x200080614 (8590460436)
  Image Width: 768 Image Length: 1024
  Tile Width: 256 Tile Length: 256
  Resolution: 392, 393 pixels/cm
  Bits/Sample: 16
  Compression Scheme: None
  Photometric Interpretation: min-is-black
  FillOrder: msb-to-lsb
  Samples/Pixel: 1
  Planar Configuration: single image plane
```

 - Bio-Formats 6.5.1 detects this file using the generic TIFF reader (with no sub-resolution) support and treats each IFD as a separate series
 - As per the above, an OMERO import workflow will create a multi-image fileset with only one resolution level for each image and the pyramid calculation will be triggered every time the image dimensions exceed the OMERO threshold.
- The pattern approach might be 

I think the options to allow the import of the data are the following:

- natively support this pyramidal PFF in Bio-Formats and OMERO. This likely requires either a new reader or a significant change to an existing reader. As [discussed in the image.sc forum](https://blog.openmicroscopy.org/community/file-formats/2019/06/25/formats/), this is unlikely to be prioritized by the open-source OME team. If anyone had capacity to look into this approach, I expect the discussion at https://github.com/ome/bioformats/issues/3300 is relevant to this use case and gives initial pointers. Shortly, the TIFF layout generated by Nanostring GeoMx is very similar to the one generated by `libvips` and has some overlaps with the custom pyramidal file format that has been used a decade ago for storing and publishing [the large Zebrafish image](https://idr.openmicroscopy.org/webclient/img_detail/4495402/?dataset=4408)
- convert this file into a pyramidal file format supported by Bio-Formats/OMERO. For OME-TIFF, the main change is that resolutions need to be stored as SubIFDs on the primary IFD rather than separate IFDs. Using the Bio-Formats Java API, it might be possible to adapt an example like [WritePyramid.java](https://github.com/ome/bioformats/blob/v6.5.1/components/formats-gpl/utils/WritePyramid.java) to read all the resolutions from the TIFF file and store them into an output OME-TIFF. An alternate workflow might be to use the https://github.com/glencoesoftware/bioformats2raw and https://github.com/glencoesoftware/raw2ometiff/ conversion pipeline. Here again some code will need to be injected to turn the individual series/images into sub-resolutions before the conversion
- for the multi-channel aspect of the data, I think the two options are either creating a single OME-TIFF file with sub-resolutions for each IFD/channel or creating one pyramidal OME-TIFF file per channel and tying them toghether via a companion file. Both approaches should be legit and there might be reasons to choose one rather than another specific to the pipeline/consumption of this data.

Coming back to the initial point, having the data generated in a supported multi-resolution multi-channel image format by the acquisition system itself is obviously the ideal solution. Are there alternative export options available?

melissalinkert, 2022-08-10:
At this point, I would expect running `bioformats2raw -series 0` on the pattern file, followed by `raw2ometiff` to be the easiest path forward for data like this. These tools are now much more mature than they were ~2 years ago when this issue was originally opened, and now successfully handle similar use cases.

Feel free to re-open if there are further issues to work out here.

closed 2022-08-10 by melissalinkert