ome/bioformats #3615: ND2 to TIFF with correct voxel depth

JimHage, 2020-09-15:
Hi,
I am trying to convert a ND2 file of a colored neuron to TIFF or v3draw with Bio-formats in ImageJ. The issue I am having is when I convert it from ND2 to TIFF the voxel depth becomes 1, but I need it to keep the original depth so I can use the image to reconstruct and analyse the neuron in Vaa3D. Can somebody help me? I am a beginner using these programs. Thanks in advance!

dgault, 2020-09-15:
Linking this issue to the ImageSc thread: https://forum.image.sc/t/nd2-to-tiff-with-correct-voxel-depth/42830/2

dgault, 2020-09-18:
Solution copied from the forum  thread:

`this is actually due to the fact that the standard TIFF tags only provide support for physical size/resolution in X and Y but there is no relevant tag for Z which causes that piece of metadata to be lost in conversion. If you export to .ome.tiff instead then this extra metadata will be retained in the OME-XML metadata and the voxel size will be correct when reimporting.`

closed 2020-09-18 by dgault