ome/bioformats #3621: BF Migration - Required Updates

dgault, 2020-10-14:
**Dependency Repos**

- [ ] jxrlib

**Downstream Repos**

- [ ] homebrew
- [ ] image-j

**Links to downloads**

- [ ] Homebrew Formulas

**Misc Updates**

- [ ] Other required updates go here

**Potential Issues which need investigating**

- [ ] FormatTools.VERSION

jburel, 2020-10-14:
Tested GitHub actions to generate and upload artifacts see https://github.com/jburel/bioformats/releases/tag/v6.6.0-m1

jburel, 2020-10-14:
https://github.com/jburel/bioformats/releases/tag/v6.6.0-m2 now have the various zip/jars and also the sha256 so it can be used in the homebrew formula:
ignore the version mismatch, this is b/c I have tagged without changing to trigger the GH action

jburel, 2020-10-14:
new tag created with shasum and md5 to match what is available under download
I have opted not to have a sha1 or md5 per file to reduce the number of attachments

jburel, 2020-10-14:
I now build on windows and ubuntu via GH actions. so we can get rid of appveyor and travis files

jburel, 2020-10-15:
@dgault https://github.com/jburel/bioformats/releases/tag/v6.6.0-m3 should have what is currently under download.o.org
Could you double check?
See comment above about sha1 and md5 https://github.com/ome/bioformats/issues/3621#issuecomment-708626796

jburel, 2020-10-15:
Now  I have example using github actions pushing artifacts to Maven central, for both gradle and maven project

jburel, 2020-10-15:
https://ci.openmicroscopy.org/job/BIOFORMATS-DEV-latest-version-bump/ only bump the version in the pom and set the value in all the pom.xml by running a python scripts
* ``STABLE_VERSION`` in ``UpgradeChecker.java`` could be removed.
* the version in the pom.xml could be set using ``mvn versions:set -DnewVersion=XXXX``

melissalinkert, 2025-03-17:
Closing as discussed in formats meeting.

closed 2025-03-17 by melissalinkert