ome/bioformats #3628: set planarconfiguration tag for bfconvert (TIFF)?

markemus, 2020-11-02:
Is there any way to set the planarconfiguration tag when saving as TIFF with bfconvert? We need our images to be "single image plane" (1) but bfconvert saves them as "separate image planes" (2).

sbesson, 2020-11-03:
@markemus a few pointers,

https://github.com/ome/bioformats/blob/c14984f78990f7d37f81e444f609125f4ec1e5b1/components/formats-bsd/src/loci/formats/out/TiffWriter.java#L405-L406

is the location where the [TIFF PlanarConfiguration tag](https://www.awaresystems.be/imaging/tiff/tifftags/planarconfiguration.html) is set. This will consume the `IFormatWriter.isInterleaved()` API.

https://github.com/ome/bioformats/blob/c14984f78990f7d37f81e444f609125f4ec1e5b1/components/bio-formats-tools/src/loci/formats/tools/ImageConverter.java#L661

is the place within the `ImageConverter` utility where the writer interleaved state is set. As you can see, some logic that transfers the interleaved state of the reader to the writer.

What type of input data are you using in your pipeline? In particular is the source data stored as interleaved TIFF?

markemus, 2020-11-05:
@sbesson thank you for pointing me to the relevant parts of the code. We're trying to write a universal bioformats -> Aperio tif converter. Unfortunately most viewers assume Aperio tifs to stored as single image planes, even though this isn't guaranteed in the specification... ah well.

I was able to get the converter working using a second copy from the bioformats extracted tiff with "tiffcp -p contig <temppath> <outpath>". Not ideal, but at least it's possible.