ome/bioformats #3629: Bfconvert of Volocity Library Doesn't Function With Output Pattern

amrosado, 2020-11-04:
I'm trying to convert a series of images from a volocity library into a bunch of tiffs representing the image series.  If I run a bfconvert:

```
bfconvert library.mvd2 output_series_%s_%n.tiff
```

The output is just one file representing the whole library despite representing the output with the appropriate pattern.

```
VolocityReader initializing library.mvd2
[Volocity Library] -> output_series_s_n.tiff [Tagged Image File Format]
```

I believe the output should be multiple files as featured in the documentation.  Clearly the pattern is being found in the output path, but the output is not be generated like I would expect.  Is there something I'm doing wrong here?  I think potentially this might be a bug.

sbesson, 2020-11-05:
Hi @amrosado, thanks for reporting, I tried reproducing your workflow with a multi-series Volocity fileset in our curated QA repository

```
build@0451602471c8:~$ /bio-formats-build/bioformats/tools/bfconvert /uod/idr/repos/curated/volocity/chris/Fig\ 4/Fig\ 4.mvd2 output_series_%s_%n.tiff
/uod/idr/repos/curated/volocity/chris/Fig 4/Fig 4.mvd2
VolocityReader initializing /uod/idr/repos/curated/volocity/chris/Fig 4/Fig 4.mvd2
[Volocity Library] -> output_series_%s_%n.tiff [Tagged Image File Format]
	Series 0: converted 3/3 planes (100%)
	Series 1: converted 3/3 planes (100%)
	Series 2: converted 3/3 planes (100%)
[done]
2.759s elapsed (10.888889+45.11111ms per plane, 2231ms overhead)
build@0451602471c8:~$ ls
output_series_0_Fig 4B.tiff  output_series_1_Fig 4C.tiff  output_series_2_Fig 4A.tiff
```

It looks like the pattern expansion worked as expected for me. From  the output of your command, it looks like the `%` symbols were lost, is it a potential problem with espacing? Have you tried double quoting the output filename?

amrosado, 2020-11-05:
Thanks for giving me some direction into what might be wrong.  I looked more into escaping in a windows command environment and found that escaping for the pattern in windows should be done by doubling the % character (https://stackoverflow.com/questions/42032953/which-symbol-is-escape-character-in-cmd).

It might be helpful for other users to remind them that escaping is different in a windows environment.

Now, I'm having a problem with with exporting files if I include the %A tag to export the timestamp the image was taken.  I'll create a different issue for that which might also impact windows users.

closed 2020-11-05 by amrosado