ome/bioformats #3631: Bfconvert Timestamp Pattern %A Fails in Windows Command Environment

amrosado, 2020-11-05:
If I run bfconvert with the following command:

```
bfconvert 20201023_DNAForceProbe.mvd2 output_tiff_%%A_%%n_%%s.tiff
```

I receive the following output:

```
VolocityReader initializing 20201023_DNAForceProbe.mvd2
[Volocity Library] -> output_tiff_%A_%n_%s.tiff [Tagged Image File Format]
Exception in thread "main" java.io.FileNotFoundException: output_tiff_2020-11-05T15:56:40_antiCD3 CD8_0.tiff (The filename, directory name, or volume label syntax is incorrect)
        at java.base/java.io.RandomAccessFile.open0(Native Method)
        at java.base/java.io.RandomAccessFile.open(RandomAccessFile.java:347)
        at java.base/java.io.RandomAccessFile.<init>(RandomAccessFile.java:261)
        at java.base/java.io.RandomAccessFile.<init>(RandomAccessFile.java:216)
        at loci.common.NIOFileHandle.<init>(NIOFileHandle.java:130)
        at loci.common.NIOFileHandle.<init>(NIOFileHandle.java:151)
        at loci.common.NIOFileHandle.<init>(NIOFileHandle.java:165)
        at loci.common.Location.getHandle(Location.java:522)
        at loci.common.Location.getHandle(Location.java:462)
        at loci.common.Location.getHandle(Location.java:443)
        at loci.common.RandomAccessOutputStream.<init>(RandomAccessOutputStream.java:66)
        at loci.formats.FormatWriter.createOutputStream(FormatWriter.java:566)
        at loci.formats.FormatWriter.setId(FormatWriter.java:401)
        at loci.formats.out.TiffWriter.setId(TiffWriter.java:163)
        at loci.formats.ImageWriter.setId(ImageWriter.java:463)
        at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:693)
        at loci.formats.tools.ImageConverter.main(ImageConverter.java:1095)

```

I do not get the same error if I don't use the %A pattern to put the series timestamp into the output filename.  I believe the problem might be related to how the colon (:) is interpreted in the path.

sbesson, 2020-11-06:
Hi @amrosado, thanks for reporting this. Similarly to https://github.com/ome/bioformats/issues/3629, I assume Windows is the environment? 

If so, I certainly understand how `:` is problematic as it is an invalid character for a Windows filename. Unfortunately, the usage of colons comes from the [IDRO 8601](https://en.wikipedia.org/wiki/ISO_8601) standard.

I see two options: 
- either simply disallowing the usage of `%A` on Windows systems
- or replacing `:` by e.g.`_`

In addition to the above, what is the rationale for encoding the acquisition time in the output file name as opposed to using a richer file format like OME-TIFF and store the information in the associated metadata?

amrosado, 2020-11-07:
I think using a richer file format like OME-TIFF and having all the associated metadata would work best.  I'll spend some time looking through the XML output I have from my library file to find all the information in metadata I need for my analysis.  If I can help make a fix preventing windows users from losing a potentially useful feature, I think this problem might originate from the loci.common.Location class. Specifically how that calling the absoluteFilePath might need a toString method.

sbesson, 2020-11-10:
> I think using a richer file format like OME-TIFF and having all the associated metadata would work best. I'll spend some time looking through the XML output I have from my library file to find all the information in metadata I need for my analysis.

If you want to export as OME-TIFF, the main change is really to use one of the extensions of the format as the output file e.g.

```
bfconvert 20201023_DNAForceProbe.mvd2 output_%%n_%%s.ome.tiff
```

> If I can help make a fix preventing windows users from losing a potentially useful feature, I think this problem might originate from the loci.common.Location class

Thanks. For reference, the `Location` class is maintained in the [ome-common-java](https://github.com/ome/ome-common-java) component. Feel free to open a RFE or directly a contribution against this repository.