ome/bioformats #3636: Opening a large (17 GB) delta vision file.

ajasja, 2020-11-18:
I am trying to open a large dv file, but get this exception when I go to import->Bio-Formats.
Smaller dv files open correctly.

```
(Fiji Is Just) ImageJ 2.1.0/1.53c; Java 1.8.0_172 [64-bit]; Linux 5.4.0-33-generic; 688MB of 266707MB (<1%)

	at loci.formats.ImageReader.getReader(ImageReader.java:202)
	at loci.plugins.in.ImportProcess.createBaseReader(ImportProcess.java:627)
	at loci.plugins.in.ImportProcess.initializeReader(ImportProcess.java:485)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:138)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:235)
	at ij.IJ.runPlugIn(IJ.java:198)
	at ij.Executer.runCommand(Executer.java:150)
	at ij.Executer.run(Executer.java:68)
	at java.lang.Thread.run(Thread.java:748)
```

If I try to open the file using the File->Open, I get 
```
(Fiji Is Just) ImageJ 2.1.0/1.53c; Java 1.8.0_172 [64-bit]; Linux 5.4.0-33-generic; 688MB of 266707MB (<1%)
 
java.lang.IllegalArgumentException: Size exceeds Integer.MAX_VALUE
	at sun.nio.ch.FileChannelImpl.map(FileChannelImpl.java:869)
	at org.scijava.text.DefaultTextService.open(DefaultTextService.java:66)
	at org.scijava.text.DefaultTextService.asHTML(DefaultTextService.java:75)
	at org.scijava.text.io.TextIOPlugin.open(TextIOPlugin.java:70)
	at org.scijava.text.io.TextIOPlugin.open(TextIOPlugin.java:48)
	at org.scijava.io.DefaultIOService.open(DefaultIOService.java:69)
	at HandleExtraFileTypes.openImage(HandleExtraFileTypes.java:528)
	at HandleExtraFileTypes.run(HandleExtraFileTypes.java:76)
	at ij.IJ.runUserPlugIn(IJ.java:235)
	at ij.IJ.runPlugIn(IJ.java:198)
	at ij.IJ.runPlugIn(IJ.java:187)
	at ij.io.Opener.openWithHandleExtraFileTypes(Opener.java:512)
	at ij.io.Opener.openUsingHandleExtraFileTypes(Opener.java:369)
	at ij.io.Opener.openImage(Opener.java:359)
	at ij.io.Opener.openImage(Opener.java:243)
	at ij.io.Opener.open(Opener.java:109)
	at ij.io.Opener.open(Opener.java:72)
	at ij.plugin.Commands.run(Commands.java:27)
	at ij.IJ.runPlugIn(IJ.java:204)
	at ij.Executer.runCommand(Executer.java:150)
	at ij.Executer.run(Executer.java:68)
	at java.lang.Thread.run(Thread.java:748)

```

Hope this is the right repo to post in.

dgault, 2020-11-18:
Hi @ajasja, are these errors coming from a custom plugin you are developing? The first (at loci.formats.ImageReader.getReader(ImageReader.java:202) would suggest that it hasn't recognised the format.

If you download the command line tools from https://www.openmicroscopy.org/bio-formats/downloads/ and run the below command do you still see errors?:
showinf /path/to/myFile.dv

ajasja, 2020-11-18:
Hi!

Running the command on the file with 50k frames
`-rw-r--r-- 1 ajasja baker 17636136960 Nov 18 09:20 08_paint_ALF201_ch2_cent_fib_blocked_004.dv`

returns 
```bash
$ ~/bin/bftools/showinf 08_paint_ALF201_ch2_cent_fib_blocked_004.dv
Exception in thread "main" loci.formats.UnknownFormatException: Unknown file format: 08_paint_ALF201_ch2_cent_fib_blocked_004.dv
        at loci.formats.ImageReader.getReader(ImageReader.java:202)
        at loci.formats.ImageReader.getFormat(ImageReader.java:162)
        at loci.formats.tools.ImageInfo.configureReaderPreInit(ImageInfo.java:433)
        at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1030)
        at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121)
```

However running the command on a smaller dv file (that was made by opening the large file in softworx and only saving 20k frames)
`-rw-r--r-- 1 ajasja baker 10496961536 Nov 17 18:58 08_paint_ALF201_ch2_cent_fib_blocked_01.dv`

returns 
```bash
$ ~/bin/bftools/showinf 08_paint_ALF201_ch2_cent_fib_blocked_01.dv
Checking file format [Deltavision]
Initializing reader
DeltavisionReader initializing 08_paint_ALF201_ch2_cent_fib_blocked_01.dv
Reading header
Populating core metadata
Reading extended header
Populating original metadata
Populating OME metadata
Reading header
Populating original metadata
Populating OME metadata
Initialization took 0.616s

Reading core metadata
filename = 08_paint_ALF201_ch2_cent_fib_blocked_01.dv
Series count = 1
Series #0 :
        Image count = 20000
        RGB = false (1)
        Interleaved = false
        Indexed = false (true color)
        Width = 512
        Height = 512
        SizeZ = 1
        SizeT = 20000
        SizeC = 1
        Tile size = 512 x 512
        Thumbnail size = 128 x 128
        Endianness = intel (little)
        Dimension order = XYZTC (certain)
        Pixel type = uint16
        Valid bits per pixel = 16
        Metadata complete = true
        Thumbnail series = false
        -----
        Plane #0 <=> Z 0, C 0, T 0
        Plane #9998 <=> Z 0, C 0, T 9998
        Plane #9999 <=> Z 0, C 0, T 9999
        Plane #10000 <=> Z 0, C 0, T 10000
        Plane #10001 <=> Z 0, C 0, T 10001
        Plane #10002 <=> Z 0, C 0, T 10002
        Plane #19999 <=> Z 0, C 0, T 19999

Reading pixel data (0-19999)
        Read 131/20000 planes (0%)
        Read 278/20000 planes (1%)
        Read 427/20000 planes (2%)
Exception in thread "main" java.lang.OutOfMemoryError: Java heap space
        at java.base/java.nio.HeapByteBuffer.<init>(HeapByteBuffer.java:64)
        at java.base/java.nio.ByteBuffer.allocate(ByteBuffer.java:360)
        at loci.common.NIOByteBufferProvider.allocateDirect(NIOByteBufferProvider.java:126)
        at loci.common.NIOByteBufferProvider.allocate(NIOByteBufferProvider.java:112)
        at loci.common.NIOFileHandle.buffer(NIOFileHandle.java:647)
        at loci.common.NIOFileHandle.read(NIOFileHandle.java:310)
        at loci.common.NIOFileHandle.read(NIOFileHandle.java:298)
        at loci.common.NIOFileHandle.read(NIOFileHandle.java:286)
        at loci.common.RandomAccessInputStream.read(RandomAccessInputStream.java:681)
        at loci.formats.FormatReader.readPlane(FormatReader.java:523)
        at loci.formats.FormatReader.readPlane(FormatReader.java:511)
        at loci.formats.in.DeltavisionReader.openBytes(DeltavisionReader.java:233)
        at loci.formats.FormatReader.openBytes(FormatReader.java:878)
        at loci.formats.ImageReader.openBytes(ImageReader.java:449)
        at loci.formats.ReaderWrapper.openBytes(ReaderWrapper.java:334)
        at loci.formats.gui.BufferedImageReader.openImage(BufferedImageReader.java:86)
        at loci.formats.tools.ImageInfo.readPixels(ImageInfo.java:821)
        at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1055)
        at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121)
```

dgault, 2020-11-19:
Yeah, the first issue suggests that Bio-Formats hasn't been able to identify the format type. Is this the only file you have which shows this error or are there any smaller examples? Either way it is likely we would need to take a look at the file to debug it further, if there are no smaller versions I will arrange ftp access for you so you can transfer the file for further testing.

The second issue should be resolved by increasing the memory available using the command `export BF_MAX_MEM=8g` (or `set BF_MAX_MEM=8g` on Windows)

ajasja, 2020-11-20:
This is the smallest file that I have that can't be opened.

I have put the file here :
https://files.ipd.uw.edu/ajasja/debug_dv/08_paint_ALF201_ch2_cent_fib_blocked_004.dv

(or if you have the high-speed educational network).
https://research-files.ipd.uw.edu/ajasja/debug_dv/08_paint_ALF201_ch2_cent_fib_blocked_004.dv

(Please let me know when you have transferred the file)
I hope that works & thank you for looking into this!

ajasja, 2020-11-24:
Hi @dgault, did you perhaps manage to download the file? I can also send just the header, probably something is wrong there...

dgault, 2020-12-02:
Hi @ajasja, sorry for the delay in getting back to this. I was able to get the file transferred successfully and will get it tested this week.

melissalinkert, 2023-08-09:
I was going to check if this is still an issue in 7.0.0, but the file still needs to be copied to `inbox` for testing. If `showinf` alone still throws `UnknownFormatException`, `showinf -format Deltavision` would determine if this is just a type checking issue, or if there are deeper problems with reading the file.

dgault, 2023-08-30:
Sample file is now at `scratch/inbox/gh-3836`

melissalinkert, 2023-08-31:
Thanks @dgault.

I do see the same error with 7.0.0, but it looks like the file is in fact corrupted. The header was either overwritten or removed, as everything that would typically indicate the image dimensions, plane count, pixel type, etc. is set to 0. Closing as there isn't really anything we can do in the reader to fix that.

@ajasja, you may have some luck recovering the images by using `File > Import > Raw...` in ImageJ. As best I could tell, these options give a reasonable result, but you may need to experiment a bit:

[image]

closed 2023-08-31 by melissalinkert

ajasja, 2023-09-01:
Thanks @melissalinkert , I'll give it a try! I'll also try to open the file with the DeltaVision software just to double check if it is in fact corrupted.