ome/bioformats #3640: Having some difficulties to save patched image in a compressed jpeg format.

Yuvi-416, 2020-11-28:
Hello everyone,
I have an NDPI image and I want to open each series in a small tile of 512*512 and want to save that image in a compressed Jpeg format.
I am following this (https://docs.openmicroscopy.org/bio-formats/5.9.1/developers/tiling.html) (https://docs.openmicroscopy.org/bio-formats/5.9.1/developers/tiling.html) (https://github.com/dgault/bio-formats-examples/blob/6cdb11e8c64566611b18f384b3a257dab5037e90/src/main/macros/jython/OverlappedTiledPyramidConversion.py) three articles to read and write and .ndpi image.

Here is the code which I passed to read and image:
(1)
import math
import bioformats
from bioformats import ImageReader, OMETiffWriter
import javabridge
from PIL import Image
from io import BytesIO

Image.MAX_IMAGE_PIXELS = None
#########################################################################
input_any1 = "/home/Downloads/DATASET/ORIGINAL-NDPI/01.ndpi"
outFile = "/home//task-1/Dataset/bio-output/"
##########################################################################
# set the tile sizes to be used
tile_SizeX = 512
tile_SizeY = 512

javabridge.start_vm(class_path=bioformats.JARS, max_heap_size="9G")

# setup reader
reader = bioformats.ImageReader(input_any1)
omeMeta = bioformats.metadatatools.createOMEXMLMetadata()
reader.setMetadataStore(omeMeta)
reader.setId(input_any1)
metadata = javabridge.jdictionary_to_string_dictionary(reader.rdr.getMetadata())
type = reader.rdr.getPixelType()

# setup writer
writer = bioformats.PyramidOMETiffWriter()
writer.setMetadataRetrieve(omeMeta)
tileSizeX = writer.setTileSizeX(tile_SizeX)
tileSizeY = writer.setTileSizeY(tile_SizeY)
writer.setId(outFile)

# read the tiles
for series in range(reader.rdr.getSeriesCount()):
    series += 1
    reader.rdr.setSeries(series)

    for image in range(reader.rdr.getImageCount()):
        width = reader.rdr.getSizeX()
        height = reader.rdr.getSizeY()

        # Determined the number of tiles to read and write
        nXTiles = int(math.floor(width / tile_SizeX))
        nYTiles = int(math.floor(height / tile_SizeY))
        if nXTiles * tile_SizeX != width:
            nXTiles = nXTiles + 1
        if nYTiles * tile_SizeY != height:
            nYTiles = nYTiles + 1

        for y in range(nYTiles):
            for x in range(nXTiles):
                # The x and y coordinates for the current tile
                tileX = x * tile_SizeX
                tileY = y * tile_SizeY
                effTileSizeX = tile_SizeX
                if (tileX + tile_SizeX) >= width:
                    effTileSizeX = width - tileX
                effTileSizeY = tile_SizeY
                if (tileY + tile_SizeY) >= height:
                    effTileSizeY = height - tileY
                # Read tiles from the input file and write them to the output OME-Tiff
                tiles_reader = reader.rdr.openBytes(image, tileX, tileY, effTileSizeX, effTileSizeY)
                print(tiles_reader)
                writer.saveBytes(image, tiles_reader, tileX, tileY, effTileSizeX, effTileSizeY)

Error:
Traceback (most recent call last):
  File "/snap/pycharm-community/219/plugins/python-ce/helpers/pydev/pydevd.py", line 1448, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/snap/pycharm-community/219/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/yuvi/PycharmProjects/task-1/VIdeo_frame_extraction/NDPI-ext/Temp-3-Bioformat.py", line 39, in <module>
    writer = bioformats.PyramidOMETiffWriter()
AttributeError: module 'bioformats' has no attribute 'PyramidOMETiffWriter'
python-BaseException

Process finished with exit code 130 (interrupted by signal 2: SIGINT)

Obviously, an error is pretty clear. But the problem is I couldn't find that function or attribute in the python-bioformats package.
And as the function name suggests (PyramidOMETiffWriter), I think it's going to write an image in a TIFF format.
But I want to save each extracted tiles in a compressed Jpeg format as I did it bellow using openslide:
(2)
img = wsi_obj.read_region(loc, img_level, patch_size).convert("RGB")
instance_byte_str_buffer = BytesIO()
img.save(instance_byte_str_buffer, "JPEG", quality=80, progressive=False)

Is there any way to save the extracted image same as above using bioformats?
And one more thing:
When I load the same image using openslide I can see the loaded image has 9 different image dimensions but when I load the same image using bioformats it only shows 6 dimensions. Why is that? How can I solve it?

And another problem, I also can't load series 0 using above (1) code, it shows bellow error:
(3)
Exception in thread "Thread-0" loci.formats.FormatException: Image plane too large. Only 2GB of data can be extracted at one time. You can work around the problem by opening the plane in tiles; for further details, see: https://docs.openmicroscopy.org/bio-formats/6.5.1/about/bug-reporting.html#common-issues-to-check
	at loci.formats.FormatReader.openBytes(FormatReader.java:872)
	at loci.formats.FormatReader.openBytes(FormatReader.java:849)
Caused by: java.lang.IllegalArgumentException: Array size too large: 186496 x 36608 x 3 x 1
	at loci.common.DataTools.safeMultiply32(DataTools.java:1286)
	at loci.common.DataTools.allocate(DataTools.java:1259)
	at loci.formats.FormatReader.openBytes(FormatReader.java:869)
	... 1 more
Traceback (most recent call last):
  File "/snap/pycharm-community/219/plugins/python-ce/helpers/pydev/pydevd.py", line 1448, in _exec
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/snap/pycharm-community/219/plugins/python-ce/helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/yuvi/PycharmProjects/task-1/VIdeo_frame_extraction/NDPI-ext/Temp-3-Bioformat.py", line 75, in <module>
    buf = reader.rdr.openBytes(image, tileX, tileY, effTileSizeX, effTileSizeY)
  File "/home/yuvi/anaconda3/envs/task-1/lib/python3.7/site-packages/javabridge/jutil.py", line 961, in method
    result = call(self.o, name, sig, *args)
  File "/home/yuvi/anaconda3/envs/task-1/lib/python3.7/site-packages/javabridge/jutil.py", line 892, in call
    result = fn(*nice_args)
  File "/home/yuvi/anaconda3/envs/task-1/lib/python3.7/site-packages/javabridge/jutil.py", line 859, in fn
    raise JavaException(x)
javabridge.jutil.JavaException: Image plane too large. Only 2GB of data can be extracted at one time. You can work around the problem by opening the plane in tiles; for further details, see: https://docs.openmicroscopy.org/bio-formats/6.5.1/about/bug-reporting.html#common-issues-to-check
python-BaseException

Isn't Ithe code (1) which is written above trying to read an image in a patch size rather than the whole image? How to solve that too?

Finally, I know this question is more related to Python-bioformats but I didn't know where I can post this question bcoz some of the code which I used where from this package. Sorry about that.

Any small help would be highly appreciated.
Thank you

Yuvi-416, 2020-11-30:
okay, 
I found the solution for some of my problem.

closed 2020-11-30 by Yuvi-416

dgault, 2020-12-01:
Hi @Yuvi-416, glad to hear that you were able to solve some of the issues. 

The first issue you have with PyramidOMETiffWriter would likely be due to an older version of Bio-Formats being used (version 6.0.0 onwards should contain the PyramidOMETiffWriter). If you instead wanted to save as JPEG you can instead change it to JPEGWriter or JPEG2000Writer.

For compression, to see what compression types are supported for a writer you can use `writer.getCompressionTypes();` and then `writer.setCompression(compression);` to set the desired compression type. However I would advise that using OME TIFF may be preferable depending on the image type and may be worth trying.

The reading code looks correct to me, so the `Image plane too large` exception is unexpected. It might be worth checking the parameters being used for the x, y, height and width for that particular call. It may be that the tiling calculations have been incorrect somehow.

The issue with the missing resolution levels I assume is the same issue mentioned on the imagesc thread (https://forum.image.sc/t/how-to-twitch-python-bioformats-code-so-that-it-will-show-all-dimensions-like-openslide-do/45950/1).

Yuvi-416, 2020-12-01:
Hi there,
Thank you for your helpful comment.
Instead of using openbytes to read an image, now I am using read function from the reader to read an image in tiles and then saving it with the help of PIL package. I think it's working as far as i understand:
```
Code:
buf1 = reader.read(zstack, rescale=False, XYWH=(tileX, tileY, effTileSizeX, effTileSizeY))
img_PIL = Image.fromarray(buf1).convert("RGB")
img_PIL.save(instance_byte_str_buffer, "JPEG", quality=80, icc_profile=img_PIL.info.get('icc_profile'), progressive=False)
```
All problem has been solved beside that last one related to dimension. 
Actually, I am trying to convert each series each stack pyramid dimension image into a multi-frame Dicom image so that I can visualize it on my WSI viewer.
But as I have already mentioned my problem (https://forum.image.sc/t/how-to-twitch-python-bioformats-code-so-that-it-will-show-all-dimensions-like-openslide-do/45950/1) over here, bcoz of some missing pyramid currently I am not getting a pyramidal series structure.
I hope you will find some solution for it.