ome/bioformats #3642: VSI: incorrect exposure times

dgault, 2020-12-03:
This issue was raised in forum thread https://forum.image.sc/t/bio-formats-exposure-time-incorrect-when-importing-vsi-files/46080

In the OME-XML the Exposure Time is the same for each plane. However the correct values can be found in the original metadata tags as shown in the images attached in the imagesc thread.

melissalinkert, 2020-12-03:
https://trello.com/c/v47N2hGC/382-vsi-incorrect-metadata-posz-and-exp-time might be related.

dgault, 2020-12-04:
A sample file showing the issue has been received in QA-29751 and the issue was tested and reproduced with BF 6.5.1

In the OME-XML the exposure time remains the same across planes:
```
<Plane ExposureTime="1.0" ExposureTimeUnit="s" PositionX="-82449.09870917775" PositionXUnit="µm" PositionY="-68454.64727965342" PositionYUnit="µm" TheC="0" TheT="0" TheZ="0"/>
<Plane ExposureTime="1.0" ExposureTimeUnit="s" PositionX="-82449.09870917775" PositionXUnit="µm" PositionY="-68454.64727965342" PositionYUnit="µm" TheC="1" TheT="0" TheZ="0"/>
<Plane ExposureTime="1.0" ExposureTimeUnit="s" PositionX="-82449.09870917775" PositionXUnit="µm" PositionY="-68454.64727965342" PositionYUnit="µm" TheC="2" TheT="0" TheZ="0"/>
```

In the original metadata the correct values are present:
```
Microscope Exposure time (microseconds) #1	150000
Microscope Exposure time (microseconds) #2	250000
Microscope Exposure time (microseconds) #3	1000000
```

PR #3897 (2022-11-18): Olympus vsi: fix exposure time handling
Exposure times can be recorded with no tag prefix or with the `Microscope ` tag prefix. Some datasets have just one or the other, some have both. The times with no prefix are preferred if present, but the `Microscope ` ones are now used if that is all that is available.

Fixes #3642.

Compare `showinf -nopix -omexml -noflat` on QA 29751 with and without this PR.

This should be safe for a patch release.
commit 5934237f (2022-11-18): Olympus vsi: fix exposure time handling

Exposure times can be recorded with no tag prefix or with the "Microscope " tag prefix.
Some datasets have just one or the other, some have both.
The times with no prefix are preferred if present, but the "Microscope " ones are now
used if that is all that is available.

Fixes #3642.
closed 2022-11-18 by dgault