ome/bioformats #3648: CZI exposure time parse issue with ORCA Flash

yfukai, 2020-12-11:
Hi, I'm enjoying the great capability of the BioFormats, thanks maintainers!

Recently I noticed that the exposure time seems to be parsed incorrectly for some Zeiss CZI files. For example, the exposure time for the following file ( https://www.dropbox.com/s/067f7t6x4z6sicr/binning1x1-exposure10ms.czi?dl=0 ) is actually 10ms, but it's parsed as 45.8 s  with bioformats. I'm wondering if that is a bug or not.

My test code is as follows (I'm currently using python-bioformats, but I suppose it's an issue for bioformats itself as the same value appears in FIJI):
```python
import bioformats as bf
import javabridge
javabridge.start_vm(class_path=bf.JARS)
import xmltodict

file="../../../camera-dark/binning1x1-exposure10ms.czi"
meta_dict=xmltodict.parse(bf.get_omexml_metadata(file))
print(meta_dict["OME"]["Image"]["Pixels"]["Plane"][0]['@ExposureTime'])
print(meta_dict["OME"]["Image"]["Pixels"]["Plane"][0]['@ExposureTimeUnit'])
# output ... 
# 45.803669172932
# s
```
I'm using
python-bioformats==4.0.0
python-javabridge==4.0.0
xmltodict==0.12.0
in Python 3.7.9, Ubuntu 18.04.3 LTS, the latest FIJI (2.1.0/1.53c with the latest BioFormats),
and the image was taken by ZEN 2.6 Pro (Blue Edition).

The following is the information for the uploaded file viewed in ZEN
[image]

Thanks!

dgault, 2020-12-16:
Hi @yfukai, thank you for raising this issue and providing a sample file. I was able to test and reproduce the issue with the newest Bio-Formats 6.6.0 release. The incorrect exposure times do indeed look like a bug here. From the metadata in the file there appears to be a number of different exposure times and Bio-Formats has used first one for each plane. Do you know if the exposure times listed below look to be correct or is 10ms the only exposure time used? The incorrect unites being displayed is likely related to existing Issue: https://github.com/ome/bioformats/issues/3560

```
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #01: 45.803669172932
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #02: 99.999999172932
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #03: 50
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #04: 140
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #05: 149.99999917293
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #06: 200
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #07: 200
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #08: 1.00367
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #09: 10
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #10: 10
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #11: 10
Experiment|AcquisitionBlock|TimeSeriesSetup|MultiTrackSetup|Track|Channel|DataGrabberSetup|CameraFrameSetup|ExposureTime #12: 10
```

As a temporary workaround you should be able to access the correct value from the global metadata with the key `Experiment|AcquisitionBlock|AcquisitionModeSetup|Camera|ExposureTime #1`. This will be present in the OME XML as below:
```
      <XMLAnnotation ID="Annotation:1149" Namespace="openmicroscopy.org/OriginalMetadata">
         <Value>
            <OriginalMetadata>
               <Key>Experiment|AcquisitionBlock|AcquisitionModeSetup|Camera|ExposureTime</Key>
               <Value>[10]</Value>
            </OriginalMetadata>
         </Value>
      </XMLAnnotation>
```

yfukai, 2020-12-17:
Hi @dgault, thank you for your response! The exposure time I used to take this image is only 10ms (in the "Bright" channel), but the values of `ExposureTime #01...` seems to be those set for each channel: DAPI, CFP, ... etc. (not used for the acquisition.) For example, it is displayed in ZEN as following:
[image]
I also appreciate your comment for the workaround.

PR #4319 (2025-06-18): Zeiss CZI: use millisecond units for exposure times attached to channels
Backported from a private PR, this is likely to fix #3648.

Expecting test failures, but want to let tests run at least once with this change and no configuration updates so that we can see exactly what needs to be evaluated.

This and #4306 likely need configuration changes in a similar scope, so will try to do the re-configuration once for both PRs.
closed 2025-06-18 by sbesson