ome/bioformats #3653: Imaris: null PixelType exception

dgault, 2020-12-16:
Issue was raised by QA-29769 and a sample file was provided which reproduces the issue with Bio-Formats 6.6.0

The stack trace for the exception is:
```
loci.formats.FormatException: Unknown pixel type: null
	at loci.formats.in.ImarisHDFReader.initFile(ImarisHDFReader.java:315)
	at loci.formats.FormatReader.setId(FormatReader.java:1392)
	at loci.formats.ImageReader.setId(ImageReader.java:849)
	at ome.formats.importer.OMEROWrapper$4.setId(OMEROWrapper.java:167)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelFiller.setId(ChannelFiller.java:223)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.ChannelSeparator.setId(ChannelSeparator.java:293)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.Memoizer.setId(Memoizer.java:662)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at ome.formats.importer.ImportCandidates.singleFile(ImportCandidates.java:427)
	at ome.formats.importer.ImportCandidates.handleFile(ImportCandidates.java:576)
	at ome.formats.importer.ImportCandidates.execute(ImportCandidates.java:384)
	at ome.formats.importer.ImportCandidates.(ImportCandidates.java:222)
	at ome.formats.importer.ImportCandidates.(ImportCandidates.java:174)
	at org.openmicroscopy.shoola.env.data.OMEROGateway.getImportCandidates(OMEROGateway.java:5965)
	at org.openmicroscopy.shoola.env.data.OmeroImageServiceImpl.importFile(OmeroImageServiceImpl.java:1113)
	at org.openmicroscopy.shoola.env.data.views.calls.ImagesImporter.importFile(ImagesImporter.java:103)
	at org.openmicroscopy.shoola.env.data.views.calls.ImagesImporter.access$000(ImagesImporter.java:49)
	at org.openmicroscopy.shoola.env.data.views.calls.ImagesImporter$1.doCall(ImagesImporter.java:127)
	at org.openmicroscopy.shoola.env.data.views.BatchCall.doStep(BatchCall.java:144)
	at org.openmicroscopy.shoola.util.concur.tasks.CompositeTask.doStep(CompositeTask.java:226)
	at org.openmicroscopy.shoola.env.data.views.CompositeBatchCall.doStep(CompositeBatchCall.java:126)
	at org.openmicroscopy.shoola.util.concur.tasks.ExecCommand.exec(ExecCommand.java:165)
	at org.openmicroscopy.shoola.util.concur.tasks.ExecCommand.run(ExecCommand.java:276)
	at org.openmicroscopy.shoola.util.concur.tasks.AsyncProcessor$Runner.run(AsyncProcessor.java:91)
	at java.lang.Thread.run(Thread.java:748)
```

sbesson, 2021-01-04:
Unsure whether this could help this problem here but https://github.com/ome/bioformats/pull/2928 originally included an extra feature to related to the pixel type detection logic in the Imaris HDF reader (see https://github.com/ome/bioformats/pull/2928/commits/00de4a997115084294de5a4bdc9befdf776653ba).

If the change was sufficient to address issue, it might be possible to reintroduce it for a minor release. The blocker back then was the addition of a new method to the `NetCDFService` interface is a breaking change. Since then, the minimal Java version was increased to 8+ and we should be able to make use of the default methods introduced in Java 8.

melissalinkert, 2024-11-06:
Investigating this a little more (in light of #4245), the file from QA 29769 is only 31416 bytes. `h5dump` confirms that there is not actually any image data present, so this exception is not indicating that there is an unsupported pixel type, but that the image simply doesn't exist.

Closing, since there really isn't much we can do about the image being missing. If anyone feels strongly that we should try to read metadata and return a blank image instead, feel free to re-open.

closed 2024-11-06 by melissalinkert