ome/bioformats #3659: Illegal reflective access by kryo

juliangilbey, 2021-01-22:
I'm not sure where I should report this issue, so please feel free to tell me if it should go somewhere else.

I'm running QuPath, and am opening lots of images using the BioFormats library to create thumbnail images.  On one image, I get the following warning:
```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.esotericsoftware.kryo.serializers.FieldSerializer (file:/home/jdg/qupath/build/dist/QuPath-0.2.3/lib/app/kryo-2.24.0.jar) to field java.util.regex.Pattern.pattern
WARNING: Please consider reporting this to the maintainers of com.esotericsoftware.kryo.serializers.FieldSerializer
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

The only use of the kryo library in QuPath is as a dependency of BioFormats.  I had a look at the kryo GitHub page, and it's currently on version 6.0.1 or so; version 2.4.0 was released in 2014.  So if it's a bug in kryo, it may well have been fixed by now.

I am happy to share my code, in case that helps. Unfortunately, though, this issue also seems to be transient: I repeated the extraction and did not get any warnings the second time around.  But my colleague, using the same code, is also sporadically getting this same warning.

sbesson, 2021-01-22:
Hi @juliangilbey, the version of `kryo` has actually been bumped to 4.0.1 in the recent Bio-Formats 6.6.0 release. This might help with the warnings seen above.
In terms of QuPath, I see that is still on Bio-Formats 6.5.1. @petebankhead @mahdilamb is this something that is on your roadmap for Qupath 0.3.x?  Happy to create a relevant issue wherever necessary if that helps.

petebankhead, 2021-01-22:
Hi @sbesson, yes indeed - we plan to always update to the latest Bio-Formats with each release. We hope 0.3.x will be available late March.

It should be possible to drag `bioformats_package.jar` onto QuPath to add it as an extension, although you might need to remove the existing jars within the QuPath installation to avoid them being used.

Ideally it would be possible for users to update Bio-Formats independently of QuPath in an easier way. I remain a bit torn in the best to make Bio-Formats available through QuPath - improvements to our current approach would be very welcome. See also
https://forum.image.sc/t/bio-formats-as-a-library-which-dependencies/36636

(Also @melvingelbard )

mahdilamb, 2021-01-22:
Hi @juliangilbey, please do let us know if updating bioformats brings stability or if there are issues.

Best,
Mahdi

juliangilbey, 2021-01-22:
(Ah, correction, kryo is now at 5.0.3. My bad.)

@petebankhead @mahdilamb We'll try this suggestion, thanks. Though now I'm confused (and this is a QuPath question, not a BioFormats one): QuPath built quite happily, including `qupath-extension-bioformats-0.2.3.jar`, and happily says it's initialising a BioFormatsImageServer, but there is no `bioformats_package.jar` present at all! So where in the build process or runtime is `bioformats_package.jar` actually obtained and/or used?

And where would I put bioformats_package.jar (on Linux)? Would it go into `build/dist/QuPath-0.2.3/lib/app/`?

Thanks!
Julian

petebankhead, 2021-01-22:
@juliangilbey QuPath is using the 'other' way [to include Bio-Formats as a library](https://docs.openmicroscopy.org/bio-formats/6.6.0/developers/java-library.html), so it is split across multiple jars (including `formats-gpl.jar`), for the [reasons discussed in the image.sc topic](https://forum.image.sc/t/bio-formats-as-a-library-which-dependencies/36636). But that does make updating it more difficult.

juliangilbey, 2021-01-22:
Hi @petebankhead Super thanks! Could I just change the `bioformatsVersion = '6.5.1'` in `build.gradle` to read `6.6.0` and then rebuild QuPath? I presume that would do the job.

petebankhead, 2021-01-22:
@juliangilbey Yup, that should work.

(From memory, dragging the jar onto QuPath to add it as an extension may well work too - I think it *does* pick up the newer version, but possibly more by luck than design... you can see the version under *Help &rarr; Installed extensions*).

juliangilbey, 2021-01-24:
Hi @petebankhead and @mahdilamb. So I rebuilt QuPath with `bioformatsVersion == '6.6.0'` (I don't know how to drag a jar onto QuPath in Linux), and that built with no problems.
Unfortunately, though, I still randomly get the same illegal reflective access operation warning (though now it refers to `kryo-4.0.2.jar`).
This time, two images (out of the 488 I processed 5 times each, so 2 out of 2440 runs of QuPath) gave the warning, neither of which was the one that gave the original warning I reported.  And of course it's not replicable, except that my colleague experienced the same thing (but again on different images).  So there is clearly some randomness involved which is affecting things.
I can live with it, but it's really weird.
Thanks!

petebankhead, 2021-03-23:
Hello again, this resurfaces more problematically in Java 16, thanks to [JEP 396: Strongly Encapsulate JDK Internals by Default](https://openjdk.java.net/jeps/396).

When opening an .ndpi file with memoization enabled, I see an error:
```
com.esotericsoftware.kryo.KryoException: java.lang.IllegalArgumentException: Unable to create serializer "com.esotericsoftware.kryo.serializers.FieldSerializer" for class: java.util.regex.Pattern
Serialization trace:
cellomicsPattern (loci.formats.in.CellomicsReader)
readers (loci.formats.ImageReader)
reader (loci.formats.DimensionSwapper)
reader (loci.formats.FileStitcher)
helper (loci.formats.in.FilePatternReader)
readers (loci.formats.ImageReader)
        at com.esotericsoftware.kryo.serializers.ObjectField.write(ObjectField.java:101)
        at com.esotericsoftware.kryo.serializers.FieldSerializer.write(FieldSerializer.java:508)
        at com.esotericsoftware.kryo.Kryo.writeClassAndObject(Kryo.java:651)
        at com.esotericsoftware.kryo.serializers.DefaultArraySerializers$ObjectArraySerializer.write(DefaultArraySerializers.java:361)
        at com.esotericsoftware.kryo.serializers.DefaultArraySerializers$ObjectArraySerializer.write(DefaultArraySerializers.java:302)
        at com.esotericsoftware.kryo.Kryo.writeObject(Kryo.java:575)
        at com.esotericsoftware.kryo.serializers.ObjectField.write(ObjectField.java:79)
        at com.esotericsoftware.kryo.serializers.FieldSerializer.write(FieldSerializer.java:508)
        at com.esotericsoftware.kryo.Kryo.writeObject(Kryo.java:575)
        at com.esotericsoftware.kryo.serializers.ObjectField.write(ObjectField.java:79)
        at com.esotericsoftware.kryo.serializers.FieldSerializer.write(FieldSerializer.java:508)
        at com.esotericsoftware.kryo.Kryo.writeObject(Kryo.java:575)
        at com.esotericsoftware.kryo.serializers.ObjectField.write(ObjectField.java:79)
        at com.esotericsoftware.kryo.serializers.FieldSerializer.write(FieldSerializer.java:508)
        at com.esotericsoftware.kryo.Kryo.writeObject(Kryo.java:575)
        at com.esotericsoftware.kryo.serializers.ObjectField.write(ObjectField.java:79)
        at com.esotericsoftware.kryo.serializers.FieldSerializer.write(FieldSerializer.java:508)
        at com.esotericsoftware.kryo.Kryo.writeClassAndObject(Kryo.java:651)
        at com.esotericsoftware.kryo.serializers.DefaultArraySerializers$ObjectArraySerializer.write(DefaultArraySerializers.java:361)
        at com.esotericsoftware.kryo.serializers.DefaultArraySerializers$ObjectArraySerializer.write(DefaultArraySerializers.java:302)
        at com.esotericsoftware.kryo.Kryo.writeObject(Kryo.java:575)
        at com.esotericsoftware.kryo.serializers.ObjectField.write(ObjectField.java:79)
        at com.esotericsoftware.kryo.serializers.FieldSerializer.write(FieldSerializer.java:508)
        at com.esotericsoftware.kryo.Kryo.writeObject(Kryo.java:557)
        at loci.formats.Memoizer$KryoDeser.saveReader(Memoizer.java:206)
        at loci.formats.Memoizer.saveMemo(Memoizer.java:968)
        at loci.formats.Memoizer.setId(Memoizer.java:697)
```

Since QuPath v0.3 will use Java 16, I've been exploring workarounds.

One option that seems to work is to pass `--add-opens java.base/java.util.regex=ALL-UNNAMED` to the JVM, but I'd really rather avoid this in the longer term. I have a sneaking suspicion that more errors will arise, and more and more packages will need to be opened in this way.

I'm not sure where `CellomicsReader` fits in, but the specific error seems related to the fact it contains a [`Pattern`](https://github.com/ome/bioformats/blob/35070be2c8ab475c357f5218e5a22185247ab135/components/formats-gpl/src/loci/formats/in/CellomicsReader.java#L88) object as a field.

Since it seems to be a cached value that will be replaced if `null`, would it be an option to simply make this `transient`? I assume (but don't know) kryo would skip it then and hopefully the warnings (from Java 11) and errors (from Java 16) would go away.

sbesson, 2021-03-24:
Hi @petebankhead thanks for the feedback. A few comments from our side:

- we are only testing Bio-Formats on LTS Java versions (currently 8 and 11) and there is limited knowledge and support for other versions. Consumers are obviously free to make their own minimal requirements and we are aware QuPath is quickly adopting new Java versions.
- feedback for Java versions above 11 is valuable anyways in the perspective of the upcoming 17 LTS release as I imagine it will give a preview some of the deprecated features 
- regarding your specific issue, the cached `Pattern` file stores information about the HCS layout as there are at least two naming variants. I don't see a compelling reason to serialize this information as a `Pattern` object. Note however, making it `transient` might have some performance impact when accessing Cellomics plate on the fly. Minimally, this should be tested.
- we just released Bio-Formats 6.6.1 and are planning for a more substantial minor release that could include breaking serialization changes. As always, we will welcome external contributions and can provide assistance with testing it on our systems using our curated QA repository.

petebankhead, 2021-03-24:
Thanks @sbesson, I spent some hours exploring this in Bio-Formats.

I can confirm that setting the `Pattern` to be `transient` overcomes the issue I was seeing with `CellomicsReader` whenever I build the jar and use it directly (although I'm not entirely sure why `CellomicsReader` features whenever I was using the `OS-2.ndpi` OpenSlide file for testing) . I checked this with a simple Java class, passing the image

```java
import java.io.IOException;

import loci.formats.FormatException;
import loci.formats.ImageReader;
import loci.formats.Memoizer;
import loci.formats.MetadataTools;
import loci.formats.meta.IMetadata;

/**
 * Simple class to confirm whether memoization causes errors in Java 16+.
 */
public class CheckForMemoizationErrors {
	
	public static void main(String[] args) {
		if ( args == null || args.length< 1 ){
			System.err.println("Name required");
			System.exit(1);
			        }
		for (String path : args) {
			try (Memoizer memoizer = new Memoizer(new ImageReader())) {
				System.out.println("Reading: " + path);
				IMetadata metadata = MetadataTools.createOMEXMLMetadata();
				memoizer.setMetadataStore(metadata);
				memoizer.setId(path);
				System.out.println("Image count: " + metadata.getImageCount());
			} catch (IOException | FormatException e) {
				e.printStackTrace();
			}
		}
	}

}
```
This works with a warning on Java 11, and fails with an error in Java 16.

However, I couldn't replicate it with a very easy unit test added to the existing `MemoizerTest`, possibly because when I run the existing Bio-Formats unit tests I see the message
```
2021-03-24 15:57:30,416 [main] ERROR loci.formats.ClassList - "loci.formats.in.CellomicsReader" is not valid.
```

Nevertheless, without me making any changes whatsoever I also saw that
```
mvn -Dtest=MemoizerTest -pl components/formats-bsd test
```
fails with Java 16. The messages are
```
2021-03-24 15:57:30,524 [main] ERROR loci.formats.Memoizer - deleting invalid memo file: /var/folders/9b/3jc1753j035dlgntdr26w3bc0000gn/T/loci.formats.utests.MemoizerTest.17708860208738848203/.test&pixelType=int8&sizeX=20&sizeY=20&sizeC=1&sizeZ=1&sizeT=1.fake.bfmemo
java.lang.IllegalAccessError: failed to access class loci.formats.in.JPEGReader$DefaultJPEGReader from class loci.formats.in.JPEGReader$DefaultJPEGReaderConstructorAccess (loci.formats.in.JPEGReader$DefaultJPEGReader is in unnamed module of loader 'app'; loci.formats.in.JPEGReader$DefaultJPEGReaderConstructorAccess is in unnamed module of loader com.esotericsoftware.reflectasm.AccessClassLoader @1ef6d34c)
	at loci.formats.in.JPEGReader$DefaultJPEGReaderConstructorAccess.newInstance(Unknown Source) ~[na:na]
	at com.esotericsoftware.kryo.Kryo$DefaultInstantiatorStrategy$1.newInstance(Kryo.java:1275) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.Kryo.newInstance(Kryo.java:1139) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.FieldSerializer.create(FieldSerializer.java:562) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.FieldSerializer.read(FieldSerializer.java:538) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.Kryo.readObject(Kryo.java:731) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.ObjectField.read(ObjectField.java:125) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.FieldSerializer.read(FieldSerializer.java:543) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.Kryo.readObject(Kryo.java:731) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.DefaultArraySerializers$ObjectArraySerializer.read(DefaultArraySerializers.java:391) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.DefaultArraySerializers$ObjectArraySerializer.read(DefaultArraySerializers.java:302) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.Kryo.readObject(Kryo.java:731) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.ObjectField.read(ObjectField.java:125) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.FieldSerializer.read(FieldSerializer.java:543) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.Kryo.readObject(Kryo.java:731) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.ObjectField.read(ObjectField.java:125) ~[kryo-4.0.2.jar:na]
	at com.esotericsoftware.kryo.serializers.FieldSerializer.read(FieldSerializer.java:543) ~[kryo-4.0.2.jar:na]
[... 125 lines cut ...]
	at loci.formats.utests.MemoizerTest.checkMemo(MemoizerTest.java:86)
	at loci.formats.utests.MemoizerTest.testConstructorTimeElapsedDirectory(MemoizerTest.java:159)

[INFO] 
[INFO] Results:
[INFO] 
[ERROR] Failures: 
[ERROR]   MemoizerTest.testConstructorTimeElapsed:132->checkMemo:86 expected [true] but found [false]
[ERROR]   MemoizerTest.testConstructorTimeElapsedDirectory:159->checkMemo:86 expected [true] but found [false]
[INFO] 
[ERROR] Tests run: 12, Failures: 2, Errors: 0, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  5.618 s
[INFO] Finished at: 2021-03-24T15:57:31Z
[INFO] ------------------------------------------------------------------------
```

The cause of the problem seems to be the same. The easy workaround in that case is to make `DefaultJPEGReader` public but I suspect that isn't desirable. It is fine in Java 11.

[... cut ...]  I'm afraid I won't have time to delve any deeper in the next month or so.

Regarding QuPath quickly adopting new Java versions, this is because Java 11 doesn't have jpackage - and we need that for creating our installers. We've resisted using other new features so our code should be compatible with Java 11. We skipped Java 15 because too much had changed, but we are aiming for Java 16 so we're ready for 17 when it comes later this year... and hopefully we can stick with LTS releases afterwards.

tlambert03, 2021-09-15:
just chiming in to say that I see this as well using bioformats 6.7.0, in python via jpype.

```
❯ java --version
openjdk 11.0.9.1 2020-11-04 LTS
OpenJDK Runtime Environment Zulu11.43+55-CA (build 11.0.9.1+1-LTS)
OpenJDK 64-Bit Server VM Zulu11.43+55-CA (build 11.0.9.1+1-LTS, mixed mode)
```
```py
import jpype
jpype.startJVM(classpath='path/to/bioformats_package.jar')
loci = jpype.JPackage("loci")
loci.common.DebugTools.setRootLevel("ERROR")
r = loci.formats.Memoizer(loci.formats.ImageReader())
r.setId('whatever.tif')
```
```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.esotericsoftware.kryo.util.UnsafeUtil (file:bioformats_package.jar) to constructor java.nio.DirectByteBuffer(long,int,java.lang.Object)
WARNING: Please consider reporting this to the maintainers of com.esotericsoftware.kryo.util.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

petebankhead, 2022-04-21:
I am experiencing this again in Java 17 (LTS) - except I haven't been able to find a way past it because the previous workaround of `--illegal-access=permit` is no longer possible due to [JEP 396](https://openjdk.java.net/jeps/396).

Using `--add-opens java.base/java.util.regex=ALL-UNNAMED` still gets me past the first problem, but I don't know how to resolve
```
java.lang.IllegalAccessError: failed to access class loci.formats.in.JPEGReader$DefaultJPEGReader from class loci.formats.in.JPEGReader$DefaultJPEGReaderConstructorAccess (loci.formats.in.JPEGReader$DefaultJPEGReader is in unnamed module of loader 'app'; loci.formats.in.JPEGReader$DefaultJPEGReaderConstructorAccess is in unnamed module of loader com.esotericsoftware.reflectasm.AccessClassLoader @68241ff8)
	at loci.formats.in.JPEGReader$DefaultJPEGReaderConstructorAccess.newInstance(Unknown Source)
	at com.esotericsoftware.kryo.Kryo$DefaultInstantiatorStrategy$1.newInstance(Kryo.java:1275)
	at com.esotericsoftware.kryo.Kryo.newInstance(Kryo.java:1139)
```

Without solving this, I don't think it's possible to use memoization on Java 17.

sbesson, 2022-04-21:
@petebankhead I just opened https://github.com/ome/bioformats/pull/3815 which includes a minimal change to the inner `DefaultJPEGReader` class that seems to fix the unit tests on Java 17 previously failing with the `IllegalAccessError`. 

The change should be tested in our CI tomorrow also including https://github.com/ome/bioformats/pull/3796 which fixes the other known Java 17 memoization issue. Would be interested to hear if that fixes your use case as well.
Both of these changes are memo-breaking so they could not be included in the Bio-Formats 6.9.1 patch release which just went out but I would expect they will be available in the follow-up minor release and allow the serialization to become functional on Java 17 environments.

petebankhead, 2022-04-21:
Thanks @sbesson that sounds good.

Would it be possible to also incorporate the change making `Pattern` in `CellomicsReader` become `transient`, as described at https://github.com/ome/bioformats/issues/3659#issuecomment-805967885 ?

Without that, there are still Java 17 problems unless the application is launched with `--add-opens java.base/java.util.regex=ALL-UNNAMED` - which I imagine many people won't do (or won't want to do).

sbesson, 2022-04-21:
By incorporating, do you mean merging #3815 and #3796 into a single PR to simplify the review?
If so, no objection to doing so from my side but I'll consult with @melissalinkert when she is online.

petebankhead, 2022-04-21:
Sorry, ignore me, I read it too quickly and missed that both are covered!

PR #3815 (2023-01-31): Java 17 support
Fixes #3659 

Java 17 LTS was released in September 2021 and in getting increasing used in the community by consumers of Bio-Formats e.g. `bioformats2raw` or `QuPath`. While there has been few reports so far indicating, this PR aims to bring full Java 17 support to Bio-Formats in the upcoming minor release, adding the required testing infrastructure and making the minimal set of code changes.

Summary of changes:

- [7ef4d15](https://github.com/ome/bioformats/pull/3815/commits/7ef4d1522ba719c4e24f851dc84cab32b34d1c52) and [f6d9c4b](https://github.com/ome/bioformats/pull/3815/co [... cut]
closed 2023-01-31 by dgault