ome/bioformats #3661: ND2: Skewed image display

dgault, 2021-02-09:
Image was raised on imagesc thread https://forum.image.sc/t/bfconvert-nd2-rgb-to-stack-of-tiff-grays-channels/48360/7
A sample file reproducing the issue with Bio-Formats 6.6.0 can be found at https://github.com/bioimage-analysis/Files/blob/bd6977bbfb5081ba74d6202b3faf5b92ae1529f2/error.nd2

Opening the file with Elements Viewer suggests that the basic metadata (dimensions, pixel type etc) are correct.
In Elements Viewer the image displays as:

[image]

With Bio-Formats the image is displaying as:

[image]

loicsauteur, 2021-08-12:
Hi @dgault

I wanted to mention that I have observed the same issue.
A few RGB images saved as .nd2 would produce that, when opened with Bio-Formats (via Fiji). Sometimes, during the same imaging session, using the same acquisition/saving procedure, only a fraction of the nd2 files would give this skewed images (I could not identify a pattern).
Re-opening them in NIS and saving them as tif or ome.tif solves this problem, as mentioned in the [forum thread above](https://forum.image.sc/t/bfconvert-nd2-rgb-to-stack-of-tiff-grays-channels/48360/7). 

FYI: 
- only observed with RGB cameras, using a Nikon DS-Ri2 and DS-Fi3 (the two RGB camera types we have)
- observed for several NIS 5-version  (5.02_b1266, 5.11.03_1373, 5.20.02_b1453 and 5.21.03_b1489)

Could you find already a solution?

I have more example nd2 files if needed and can provide you a download link.

Thanks and bests.

dgault, 2021-08-12:
Hi @loicsauteur, thank you for the additional details, unfortunately we do not yet have a solution for this particular bug. If you have any sample files showing the issue it would always be helpful.

loicsauteur, 2021-08-12:
Please find 9 nd2 files with following link (will expire):
https://filesender.switch.ch/filesender2/?s=download&token=d658ddd7-027a-44f8-8a85-29bc328d0101
8 of them are opening showing problems, one of them is opening normally.
The biggest one might be a different problem... probably the only time I have seen that...

dgault, 2021-08-17:
Thanks @loicsauteur for the sample images, they are a great help. With the latest Bio-Formats (6.6.1) I am seeing the skewing issue pretty clearly in 6 of them, however `HistologzNikonMPVIVO_M2S2-75-25_HE_R2.3.nd2` and `HistologzNikonMPVIVO_M1S2-25-75_HE_3.nd2` appear to open correctly. Are you seeing the same issue with those 2 files or is it a different problem?

loicsauteur, 2021-08-17:
Yes, 6 of them have the described skewing error.

`HistologzNikonMPVIVO_M2S2-75-25_HE_R2.3.nd2` is one that opens correctly. It is from a same acquisition session as some of the other images. May serve you as control...

`HistologzNikonMPVIVO_M1S2-25-75_HE_3.nd2` opens almost correctly. The bottom part is screwed up... (see FIJI vs. NIS comparison below). Probably a different problem and maybe rather related to its size (and number of pixels, coming close to Java's limitation?). This problem is very rare, I observed it maybe 2-3 times. But I thought it may also be related...

[image]

dgault, 2024-07-23:
An additional report of this issue with 5 sample files showing the same skewed displayed can be found at `/uod/idr/scratch/panama_nd2 `

4 of the files show the same issue and can be reproduced with Bio-Formats 7.3.1. The last file ending `DIC_04T.nd2` appears to be correct. Changing the nd2 options for chunkmapping does not have any impact on the result.

odinsbane, 2025-05-16:
This appears to be caused by a bug in the [ND2Reader](https://github.com/ome/bioformats/blob/6a2ff9b9b2d4387b1c13b7f03d6173f1c1438498/components/formats-gpl/src/loci/formats/in/ND2Reader.java#L322)

    long post = (long) pixel * (getSizeX() - w - x) + (scanlinePad * bpp);

bpp is 1 probably because the channels are "split" but the actual bytes per pixel being read from the scan line are `pixel` sized. Changing the above line to

    long post = long post = (long) pixel * (getSizeX() - w - x) + (scanlinePad * pixel);

Fixes the problem on an example image I have.

odinsbane, 2025-05-16:
I checked the error [error.nd2](https://github.com/bioimage-analysis/Files/blob/bd6977bbfb5081ba74d6202b3faf5b92ae1529f2/error.nd2) and it turns out scanlinePad is wrong in that case. The actual padded value should be 2. 

In the [form post example](https://forum.image.sc/t/nd2-does-not-open-in-fiji-bio-formats-8-1-1/112396/3) I used the padded value should be 3.

From this forum, https://github.com/orgs/Blaze-Stars/discussions/45 it says that a scanline should be a multiple of 4 bytes. **If that is true** a "line" is `pixel*getSizeX()` we can calculate the true pad size by.

    long linewidth = pixel*getSizeX();
    long post = linewidth%4;
    if(post != 0) post = 4 - post;

That works for both error.nd2 and the forum post.

melissalinkert, 2025-05-30:
Fixed by https://github.com/ome/bioformats/pull/4309

closed 2025-05-30 by melissalinkert