ome/bioformats #3669: Issue with FV1000Reader

dominikl, 2021-03-19:
See [QA 30058](https://www.openmicroscopy.org/qa2/qa/feedback/30058/) .
Unfortuantely I couldn't find out any more details. Bioformats tool `showinf` opens the file, no errors (although doesn't show any metadata as far as I can tell). But importing it into OMERO fails.

sbesson, 2021-03-19:
It looks like the exception is triggered by the OME metadata population which is turned on when importing into OMERO:

- `showinf -nopix Image_with_failed_import.oib` returns without issue
- `showinf -nopix -omexml Image_with_failed_import.oib` fails with the same exception reported by the import log
   ```
    Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 1, Size: 0
	at java.util.ArrayList.rangeCheck(ArrayList.java:657)
	at java.util.ArrayList.get(ArrayList.java:433)
	at ome.xml.model.Instrument.getFilter(Instrument.java:471)
	at ome.xml.meta.OMEXMLMetadataImpl.setFilterID(OMEXMLMetadataImpl.java:7156)
	at ome.xml.meta.FilterMetadata.setFilterID(FilterMetadata.java:994)
	at loci.formats.in.FV1000Reader.populateMetadataStore(FV1000Reader.java:1096)
	at loci.formats.in.FV1000Reader.initFile(FV1000Reader.java:992)
	at loci.formats.FormatReader.setId(FormatReader.java:1421)
	at loci.formats.ImageReader.setId(ImageReader.java:849)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:650)
	at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1035)
	at loci.formats.tools.ImageInfo.main(ImageInfo.java:1121)
  ```

In particular this section of the code might need to be reviewed

https://github.com/ome/bioformats/blob/442c35a834a8b4a1e67925d9243bdf631e8628c8/components/formats-gpl/src/loci/formats/in/FV1000Reader.java#L1094-L1097

pwalczysko, 2021-03-31:
Note: additional data submitted, and stored at 
``idr0-slot3:/uod/idr/scratch/inbox/from_Freising``

These data include 2 .oib images. More data will be incoming from the user.

cc @sbesson @dominikl @dgault

PR #3684 (2021-04-15): FV1000Reader: do not assume channels and filters have matching indexes
Fixes #3669 
 
Since the creation of the EmissionFilter metadata is conditional to the
barrierFilter value, this commit ensures a filterIndex variable is initialized
and incremented independently of the channelIndex variable.

Using the sample files uploaded as part of the corresponding issue and either `showinf -nopix  -omexml` or `omero import`:

- one file should fail and the other should read successfully without this PR
- both files should be successfully read with this PR
closed 2021-04-15 by dgault