ome/bioformats #3678: Ventana BIF: Z stacks not supported

dgault, 2021-03-31:
Issue was raised on imagesc thread https://forum.image.sc/t/qupath-not-opening-bif-z-stacks/50674
A sample file along with a related `tifp` file have been provided which reproduce the issue (tested with BF 6.6.1)

Currently the sample file opens with 11 series (and internally 11 ifds are parsed) however the series are supposed to contain 15 z slices.

dgault, 2022-05-18:
A further mention of this issue on image-sc: https://forum.image.sc/t/help-qupath-not-showing-tif-bif-ventana-z-stacks/67136

dgault, 2022-05-18:
Link to a whitepaper that documenst the BIF format: https://diagnostics.roche.com/content/dam/diagnostics/Blueprint/en/pdf/rmd/Roche-Digital-Pathology-BIF-Whitepaper.pdf

jimtalksdata, 2023-04-12:
Dug a little into this.

In the tif file, the image data is defined in 4 dimensions, as follows: 

print(z_image.shape)
(15, 14336, 12288, 3)

Here is a proof of concept python script to do the actual image extraction into single-level TIFs (you'll need `tifffile` and `imagecodecs` installed):

```
import tifffile

# Load the image
input_file = 'zstack.tif'

with tifffile.TiffFile(input_file) as tif:
    # Extract the image with all z-layers
    z_image = tif.asarray(series=0)

# Get the number of z-layers
z_layers = z_image.shape[0]

# Loop through z-stacks and save each one as a separate file
for z in range(z_layers):
    print("Saving z-stack " + str(z) + " of " + str(z_layers))
    # Extract the z-layer image
    single_z_image = z_image[z]

    # Save the z-layer image as a separate file
    output_file = f"output_zstack_{z}.tif"
    tifffile.imwrite(output_file, single_z_image)

print("All z-stacks have been saved as separate files.")
```

dgault, 2023-04-13:
Thanks @jimtalksdata for investigating, that is useful to know!