ome/bioformats #3683: java.io.IOException: block[ 0 ] already removed #386

antonkulaga, 2021-04-09:
I am often getting "IOException: block[ 0 ] already removed  " errors for zvi files, here is a example file that causes it https://drive.google.com/file/d/1U-d2MApVq3GROoEyWTPSA_Uz03AVA2lU/view?usp=sharing
I use bioformats through pims (so also opened an issue https://github.com/soft-matter/pims/issues/386 , but I also saw it in other libs that use bioformats under the hood )

dgault, 2021-04-09:
Hi @antonkulaga, thank you for opening the issue and providing a sample file. I have not been able to reproduce the issue when using Bio-Formats directly. Would you have a short piece of example code showing how you are accessing the images via pims?

Do you have Also do you know what other applications or libraries you have seen the issue with? Was it the same error in these cases?

antonkulaga, 2021-04-13:
I use pims python library that is using latest bioformatcs under the hood (I also tried older versions without success).
Here is the code https://github.com/antonkulaga/microscope_video_conversion
The files that cause the exception are https://drive.google.com/drive/folders/1ftBycWC-vff9nDTuB79uRTw2Vf714VvH?usp=sharing
Here is the screenshot of the error log 
[image]

dgault, 2021-04-13:
Thanks @antonkulaga, I will try to reproduce with the samples provided and update this issue once I've learnt more

dgault, 2021-04-14:
As a quick update, I can confirm that the issue is reproducible directly with Bio-Formats 6.6.1 using the sample files provided