ome/bioformats #3696: Relicense JHDFService (and CellH5Reader/Writer)

sbesson, 2021-05-20:
The reader for the BigDataViewer file format was introduced in https://github.com/ome/bioformats/pull/3151 in the context of the academic BBSRC BB/P027032/1 grant. Given the funding, the provenance and the academic nature of the format, there is no rationale for having this format under the GPL-2 license and the @ome/formats team has discussed relicensing it under the permissive BSD 2-clause license.

To access the HDF5 binary data, this reader depends on the JHDF service, also licensed as GPL-2, introduced as part of the work adding support for the CellH5 file format - see https://github.com/ome/bioformats/pull/1637, 

@sommerc as the original author and primary contributor, would you be willing to relicense these classes under the more permissive BSD-2 clause?

-  components/formats-gpl/src/loci/formats/services/JHDFService.java
- components/formats-gpl/test/loci/formats/utests/JHDFServiceTest.java
- components/formats-gpl/src/loci/formats/services/JHDFServiceImpl.java
- components/formats-gpl/test/loci/formats/utests/MissingJHDFServiceTest.java

/cc @melissalinkert

sommerc, 2021-05-21:
Absolutely! This makes perfect sense for both. 

Cheers, Chris

sbesson, 2021-05-21:
Fantastic, thanks @sommerc for getting back to us. Looking at the contributor list for the concerned files, the only other two authors are @melissalinkert and myself.  

I authorize my contributions to the JHDFServiceImpl.java to be relicensed as BSD-2.

Mostly for bookkeeping purposes, @melissalinkert I assume there is no objection from your side?

PR #3700 (2021-07-13): Relicense JHDF service and CellH5 readers under BSD-2
Fixes #3696

- migrates  the `JHDFService` interface and its implementation as well as the `CellH5` reader/writer classes under `formats-bsd` and updates the headers of these classes 
- moves the `jhdf5` dependency under `formats-bsd`
- migrates the `JHDFServiceTest` and `MissingJHDFServiceTest` tests, adds `JHDFServiceTest` to the default testng suite and migrate the `test-no-jhdf` target under `formats-bsd`
- adds a `test-no-jhdf` execution to the Maven test phase of `formats-bsd` to cover the `MissingJHDFServiceTest` as part of the Maven [... cut]
closed 2021-07-13 by dgault