ome/bioformats #3697: OBF: Slow OME XML validation

ngladitz, 2021-05-21:
I've been forwarded https://forum.image.sc/t/abberior-files-slow-to-open/51273
which pointed out that the OBF reader takes a very long time in `OMEXMLServiceImpl.validateOMEXML` mostly in `java.net.SocketInputStream.socketRead0`. Perhaps downloading schemas?

Other readers don't seem to be doing this so I tried to check when / why this was added for OBF and found:
"Update OME-XML parsing to include a validation check" (1160987eef56c73e3ca832210f3e88f0dd5c0160)

I am unsure what the original issues was for which the validation was added or how validation fixes it.
Presumably it has something to do with old software versions generating bad metadata that had to be coerced with some heuristic.
Assuming anyone can reconstruct the original issue ... would it be possible to work around it without doing the full validation run?
Alternatively make it conditional? Not entirely sure which criteria would work though.

dgault, 2021-05-24:
When I last profiled the OBF files it was indeed the downloading of the schema for validation which caused the slowness you are seeing. The validation was added to handle invalid XML which was causing exceptions in the files. There is an existing metadata option for toggling validation, this could potentially be reused in this reader. The downside of disabling it is that it could lead to invalid files being unable to be read. But that may be one option.

When you currently read your files do you see a warning in the logging such as below?:
`Could not parse OME-XML metadata`

ngladitz, 2021-05-24:
@dgault Thank you for looking into this. There should not be any validation errors with files generated by current software versions.
No 100% guarantee but I've added tests that use Bioformats' OME XML validation in CI runs of one of our OBF generating products around 2017 and I think we've got something similar for the other products as well. If there are issues they should preferably be fixed in the products that generate the files.

sbesson, 2021-05-24:
Another relevant link was https://github.com/ome/bioformats/pull/3268 where I had tried to fix the offline schema validation by using the internal OME schemas but was still hitting a bottleneck when trying to perform an online validation.

Following the original investigation started on the image.sc thread, I eventually traced the limiting calls to https://www.w3.org/Help/Webmaster#slowdtd i.e. the fact the W3C web servers are intentionally throttling the remote calls. Comparing with the OME schema access time:

```
$ time curl -O https://www.openmicroscopy.org/Schemas/OME/2016-06/ome.xsd
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  255k  100  255k    0     0   203k      0  0:00:01  0:00:01 --:--:--  203k

real	0m2.635s
user	0m0.026s
sys	0m0.027s
$ time curl -O http://www.w3.org/2001/xml.xsd
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  8836  100  8836    0     0    578      0  0:00:15  0:00:15 --:--:--  2196

real	0m15.278s
user	0m0.011s
sys	0m0.014s
```

I see two potential separate ways to approach this problem:

- is XML validation a requirement for this reader? As mentioned above, validation could be disabled either via an option or maybe based on another criteria like the detected format version
- can the performance of the XML validation be improved? This probably requires additional work but IMHO this is the most valuable long-term fix and the beyond the scope of this reader. Assuming a largely instantaneous process, there should be no cost to validating metadata more systematically. I suspect this would require additional caching at the ome-model level. Depending on the direction we want to take here, happy to create a separate issue on this repository.

ngladitz, 2021-05-25:
Hey @sbesson, thank you for looking at this as well.
I would like to see both suggested approaches implemented independently.

The second approach would presumably speed up my CI runs as well (I used to run a separate validation process for each test file which ran several minutes; now I use a single process so the schema only has to get loaded once which is quite a bit faster (less than half a minute) but it still doesn't feel as fast as it could / should be).

As for the first approach ...

From what I remember you do have a database of (non public) test files which you test against and I assume irregardless of the chosen issue fix those test files (which probably contain a mix of files with both good and bad OME meta data) would still have to (and probably should) keep passing / working.

Having validation opt-out would mean the reader stays slow by default even when new files don't necessarily benefit; discovery and use of the option might be problematic depending on context.

Having validation opt-in would mean legacy files with broken metadata might also break by default.
I am unsure how exactly it would break though; i.e. if / how fatal the resulting errors would be and if the files could still be worked with at some level.

File format version does not directly correspond with the metadata we write so it might not be the best heuristic to detect if validation should be done. OME Creator might work but it is unstructured / free textish and might be difficult to get right (include all products / variants) / could break easily (when products get added /renamed / reformatted).

Any chance you could share just the metadata of the OBF files in your database and / or try taking out validation and report back if / how the existing test files fail and if there seems to be any clear / usable attribute values one could use to conditionally disable validation?

sbesson, 2021-05-26:
@ngladitz thanks for the additional thoughts. I will capture the validation performance issue separately, on the ome-model repository

For the OBF datasets, apart from the ones you shared with us as part #3528 , all our samples had been shared privately. This does not mean that an extract of the relevant metadata cannot be shared but we would need to review them individually before doing so.

As a first step, feel free to open an  pull request to disable the validation so that we can investigate its impact on our curated QA repository tests. Once we have narrowed down the failing datasets, we can certainly review whether the metadata or extracts of it can be shared.

ngladitz, 2021-05-26:
@sbesson thanks I've created the corresponding MR #3698.
Not a proper benchmark in any way but the simplest image files on my hardware now anecdotally load in 0.2s instead of 16s.
The difficult part will probably be how to disable this conditionally but the potential outcome certainly looks promising.

dgault, 2021-07-01:
Closing this issue as the PR https://github.com/ome/bioformats/pull/3701 provides a solution

closed 2021-07-01 by dgault