ome/bioformats #3699: Use the cached XML schema for improved validation

sbesson, 2021-05-27:
_Originally posted by @sbesson in https://github.com/ome/bioformats/issues/3697#issuecomment-847306437_

The OME-XML validation calls used by certain readers and the utilities try to retrieve the upstream XML schema from the W3C website which is throttling such calls - see https://www.w3.org/Help/Webmaster#slowdtd. For libraries, the accepted recommendation is to cache these schemas and use catalog files.

The ome-model repository already includes a [copy of the XML schema](https://github.com/ome/ome-model/blob/v6.2.2/specification/src/main/resources/released-schema/external/xml.xsd) as well as [a catalog file](https://github.com/ome/ome-model/blob/master/specification/src/main/resources/released-schema/external/catalog.xml) and these resources are bundled as part of the `org.openmciroscopy:specification` JAR but it looks like these are not being used. 

Being able to use the internal cache file should greatly improve the experience and performance when it comes to validating OME-XML fragments.

ngladitz, 2021-05-27:
Don't quite know what I am doing here or if it is generic enough or how it is intended exactly but this does seem to work:
```patch
diff --git a/components/formats-api/src/loci/formats/services/OMEXMLServiceImpl.java b/components/formats-api/src/loci/formats/services/OMEXMLServiceImpl.java
index 2ba3a7069a..a20d7047a5 100644
--- a/components/formats-api/src/loci/formats/services/OMEXMLServiceImpl.java
+++ b/components/formats-api/src/loci/formats/services/OMEXMLServiceImpl.java
@@ -174,6 +174,8 @@ public class OMEXMLServiceImpl extends AbstractService implements OMEXMLService
             /* from specification.jar */
             return getClass().getResourceAsStream("/released-schema/" +
                  matcher.group(1) + "/" + matcher.group(2) + ".xsd");
+          } else if(url.equals("http://www.w3.org/2001/xml.xsd")) {
+            return getClass().getResourceAsStream("/released-schema/external/xml.xsd");
           } else {
             return null;
           }
```

Got me the sameish speed improvement as removing validation in the OBF reader for my small test file.
This may indicate that removing validation (especially if it turns out to be problematic after all) can perhaps be omitted but I've still got to test this with something more bulky.

dgault, 2021-07-01:
Closing this issue as the PR https://github.com/ome/bioformats/pull/3701 provides a solution

closed 2021-07-01 by dgault